MikePopoloski/slang #659: coverpoint errors about undeclared identifiers which exist in parent class

jrudess, 2022-11-18:
I'm seeing failures with recent builds on some covergroups/coverpoints of the form:

```systemverilog
file.sv:84:23 error: use of undeclared identifier 'var'
    label: coverpoint var;
                      ^
```

I haven't had success in creating a standalone example that reproduces the error.   'var' is a member variable of the parent class that the covergroup is defined within.  Can you describe the change from 9b8aec0c2b9f9ab3b58a60e90663f0bd85e94019?

MikePopoloski, 2022-11-18:
I think that diff is unlikely to be the problem, that one is (very slightly) changing how dotted member access of a covergroup variable operates. More likely is this one: 0164d4465eedb9ee221699f438b4a3942d0c5634 which did a bit of reorganization of how cover crosses resolve their types. Perhaps the bug introduced relates to implicit cross symbols?

jrudess, 2022-11-18:
Confirmed that issue starts with 0164d4465eedb9ee221699f438b4a3942d0c5634

MikePopoloski, 2022-11-18:
OK yeah, definitely something funky going on with covergroups declared inside classes; they are "implicit" and a property gets created with their name instead. The implicit covergroup wasn't being added to the parent class scope so some lookup stuff was broken. Fix is here: f9812445984f0c12f315462bac73facdf337a4b7

I don't know if it fixes your original issue, but it definitely fixes *an* issue.

jrudess, 2022-11-18:
That resolves the 'undeclared identifier' errors, but got a new one related to accessing the special covergroup options (LRM Table 19-1).

```systemverilog
class txn;

    covergroup cg_txn;
        option.comment = "txn cov description";
    endgroup

    function new();
        cg_txn = new();
    endfunction

endclass
```
```systemverilog
slangtest96.sv:4:9: error: invalid use of non-static class property 'option' from nested class 'txn'
        option.comment = "Txn coverage";
        ^~~~~~~~~~~~~~
warning: no top-level modules found in design [-Wmissing-top]
```

MikePopoloski, 2022-11-19:
Fixed that in f413d9d51f1c2e63f8bf1c4069e7d23712605589

These have been broken since I first implemented covergroups. I'm not sure why my recent commit uncovered it but at least slang is actually checking the validity of covergroups declared inside classes now.

jrudess, 2022-11-19:
All issues I was seeing are resolved.

closed 2022-11-19 by jrudess