MikePopoloski/slang #666: module XMR: using module name instead of instance name

marcelchangTW, 2022-11-23:
**Describe the bug**

When using module name instead of instance name, slang reports an error: could not resolve hierarchical path.
It should be a module XMR case and it is allowed in LRM, right?

**To Reproduce**

`module top();
sub s();
initial begin
	sub.r[0:1]=1;
end
endmodule

module sub();
	reg[0:1]r;
	wire[0:1] b;
	assign b=r;
endmodule`

this case can hit this issue, error: could not resolve hierarchical path name 'sub'.
Will slang support this kind of usage?

MikePopoloski, 2022-11-23:
Hmm, I'm not sure if this is actually allowed or not. I see that VCS allows it, but four other tools I tried (including Xcelium and ModelSim) do not. From the LRM it's not clear whether this should be allowed -- upward names seem to always be described in terms of modules along the instantiation path and not any other instance that happens to be in scope, but again the wording is not clear. I assume you have some large design that relies on this?

This would definitely work everywhere if you changed `sub.r[0:1] = 1` to be `s.r[0:1] = 1` as then it's a regular downward hierarchical reference.

marcelchangTW, 2022-11-24:
Thank Mike! You are right. Only upward name referencing is defined in LRM, not clear definition for this case. I will close this.

closed 2022-11-24 by marcelchangTW