MikePopoloski/slang #668: External names for unnamed generate blocks

marcelchangTW, 2022-11-25:
**Describe the bug**
I tried the example in LRM 27.6  external names for unnamed generate blocks, but I can't access the implicit name genblk1.a, and don't see any 'genblk1' information in ast json file.

**To Reproduce**
Using the example in LRM 27.6 and put the following check code into top module.
```
initial begin
    if(genblk1.a == 1'b0)
        $display("Passed");
    else
        $display("Failed");
end
```
it will get an error: use of undeclared identifier 'genblk1'; did you mean 'genblk2'?

MikePopoloski, 2022-11-25:
Unnamed generate blocks cannot be referenced from within SystemVerilog code. The implicit name is for external tools, API access, etc. If you want to access the members you need to give the block a name.

If you want this external name to be visible in the JSON that's doable.

marcelchangTW, 2022-11-28:
OK, I understood. I close it first. Thanks!

closed 2022-11-28 by marcelchangTW