MikePopoloski/slang #675: Error when indexing string-map with a string variable

jrudess, 2022-11-29:
Beginning with af62de2cb678eedb4e45570151a5ad8d215c1389 I'm seeing a weird case that fails on my build of slang on work-machine, but the reproducer passes on my build of slang on my home machine.  Both were built with `-DCMAKE_BUILD_TYPE=Release`.

**Failing Build:**
gcc 12.1.0
cmake 3.20.5
Using remote fmt

**Passing Build:**
gcc 12.2.0
cmake 3.25.5
Using system fmt

```
slangtest101.sv:7:31: error: expression type 'string' is not integral
            e = enum_type_map[str];
                              ^~~
```
```Systemverilog
virtual class C #(type enum_t = int);
    static function enum_t str2enum(string str);
        static enum_t enum_type_map[string];
        enum_t e;

        if (enum_type_map.exists(str)) begin
            e = enum_type_map[str];
        end else begin
            e = e.first();
        end

        return e;
    endfunction
endclass
```

jrudess, 2022-11-30:
Something odd was going on with my clone of the repo on home-machine.  Removed that directory and re-cloned and now the reproducer fails there also.

MikePopoloski, 2022-11-30:
Fixed by 9583858ac2567e319711dc49290525cf198c5498

closed 2022-11-30 by MikePopoloski