MikePopoloski/slang #685: Arguments of extern class functions treated as unused

jrudess, 2022-11-30:
```
slangtest110.sv:2:32: warning: unused formal argument 'a' [-Wunused-argument]
    extern function void f(int a);
                               ^
```
```
class C;
    extern function void f(int a);
endclass

function void C::f(int a);
    if (a) begin
        $display("a = %0d", a);
    end
endfunction
```

jrudess, 2022-12-02:
The DPI imported function case may be the same issue also.
```
slangtest114.sv:1:52: warning: unused formal argument 'env_name' [-Wunused-argument]
import "DPI-C" function string getenv(input string env_name);
                                                   ^
```
```systemverilog
import "DPI-C" function string getenv(input string env_name);

class C;
    function void f();
        $display("%s", getenv("VCS_HOME"));
    endfunction
endclass
```

MikePopoloski, 2022-12-03:
Fixed in 8a6f40b62ed1b53b8ecb4a961fe3b6e93ea11436

closed 2022-12-03 by MikePopoloski