MikePopoloski/slang #703: covergroup is triggered but treated as unused

jrudess, 2022-12-10:
I'm not too familiar with covergroups from practical usage, so let me know if you think this warning is good.

The covergroup is setup to get triggered on an event.   If the covergroup is created in-line at its definition, it gets an 'unused variable' warning.   If the create() function is un-commented, it gets an 'assigned but never read' warning.

Should the covergroup be considered as 'used' if its trigger has a driver without needing a manual cg.sample() call?

```
slangtest127.sv:11:8: warning: unused variable 'cov' [-Wunused-variable]
    cg cov = new();
       ^
```
```systemverilog
interface I;
    logic clk;

    logic a;
    event trig;

    covergroup cg @trig;
        a: coverpoint a;
    endgroup

    cg cov = new();

    always @(a) begin
       ->>trig;
    end

    // function create();
    //     cov = new();
    // endfunction

endinterface

module top;
    I i();

    assign i.a = 0;
    initial begin
        i.clk = 0;
        forever begin
            #1ns;
            i.clk = ~i.clk;
        end
    end
endmodule
```

MikePopoloski, 2022-12-11:
Probably classes as well actually, as constructing a class could involve side effects that essentially make it "used.

jrudess, 2022-12-11:
The warnings have definitely helped find dead code on class handles.  The two main cases where I think the warnings have been useful for trimming out dead code are with functions/tasks that take a handle but don't use the handle, and local function/task handles that are neither assigned or constructed.

MikePopoloski, 2022-12-13:
Fixed in b65056e3b3c70821075e12cf4855fbb3c508bd1f

closed 2022-12-13 by MikePopoloski