MikePopoloski/slang #710: Virtual interface signal accesses disallowed by classes contained in a package

jrudess, 2022-12-17:
Getting a new error that prevents virtual interfaces from being used by classes within packages.   The following example passes if the package wrapper is commented out.

```
slangtest130.sv:10:27: error: cannot use hierarchical reference from within a package
                @(posedge intf.clk);
                          ^~~~~~~~
```
```Systemverilog
interface Bus;
    logic clk;
endinterface

package P;
    class BFM;
        virtual Bus intf;
        task drive_txn();
            forever begin
                @(posedge intf.clk);
            end
        endtask
    endclass
endpackage
```

MikePopoloski, 2022-12-17:
Fixed in 6b969b45dd9ca97b676f75c8cfb94052a93f4998

jrudess, 2022-12-17:
Looks like there's a secondary issue when clocking blocks are also involved.

```
slangtest131.sv:15:17: error: cannot use hierarchical reference from within a package
                intf.cb.a <= 1'b1;
                ^~~~~~~~~
```
```systemverilog
interface Bus;
    logic clk;
    logic a;
    clocking cb @(posedge clk);
        output a;
    endclocking
endinterface

package P;
    class BFM;
        virtual Bus intf;
        task drive_txn();
            forever begin
                @(intf.cb);
                intf.cb.a <= 1'b1;
            end
        endtask
    endclass
endpackage
```

MikePopoloski, 2022-12-21:
Fixed by 2496fc35b8ca1fb73aaab98c635b72162fa3df0a

closed 2022-12-21 by MikePopoloski