MikePopoloski/slang #722: the line number in ast.josn and lint check

KingkingLiu, 2023-02-10:
I use the option "--ast-json" to generate the AST file, then we do some check on the basis of AST. However, we can't get the location of  source code for node of AST. Because of the lint check, I noticed that in the source code you compile Testcases for various diagnostics, and I was wondering if you have developed those diagnostics (or lint check).How do I use these code diagnostics, is there an option to enable diagnostics. I am looking forward to your reply, which will be of great help to me.

MikePopoloski, 2023-02-14:
The AST dump does not include line numbers, but it could do so. I don't have time to add such a feature myself but would accept a PR to do so.

I'm not able to follow the rest of your question. Yes, slang has many diagnostics and there is much documentation about how they work and how to control them on the website: https://sv-lang.com/

MikePopoloski, 2023-03-04:
Closing this -- feel free to reopen and provide more clarification if necessary.

closed 2023-03-04 by MikePopoloski