MikePopoloski/slang #724: Covergroup identification

GiamboVuolo, 2023-02-16:
### Discussed in https://github.com/MikePopoloski/slang/discussions/723

<div type='discussions-op-text'>

<sup>Originally posted by **GiamboVuolo** February 15, 2023</sup>
Hello,
I'm having trouble when getting out coverage groups informations in the ast. The fact is that when I declare and instantiate the coverage group in a module, it is shown in the ast as "kind: CovergroupType" with its coverpoints and bins, while if I instantiate it inside of a class, it sees it as "kind: ClassProperty". As consequence, the information about its bins and coverpoints is lost.
I provide a dummy test with the two configurations. 
Let me know if you can help me, thank you!

```
module test_top;

    `include "packet.sv"

    packet pkt;
    bit ok;
    bit clk = 0;

    always #10 clk = ~clk;

    covergroup cg_good @(posedge clk);
        pluto : coverpoint pkt.length   {
            bins a = {[0:25]};
            bins b = {27};
        }
        topolino : coverpoint pkt.address   {
            bins a = {1};
            bins b = {3};
        }
    endgroup

    initial begin
        cg_good cg_g= new();
        for (int i = 0; i < 2; i++) begin
            @(posedge clk);
            pkt = new($sformatf("packet_%0d", i + 1));
            ok = pkt.randomize();
            pkt.sample();
            pkt.print();
        end
        $finish();
    end

endmodule
```

```
class packet;

    string name;
    rand bit[5:0]       length;
    rand bit[1:0]       address;

    covergroup cg_bad with function sample();
        pippo : coverpoint length   {
            bins a = {6'b000000};
            bins b = {2,16};
        }
        lello : coverpoint address  {
            bins a = {2'b00};
            bins b = {2};
        }
    endgroup

    function new(string name = "CIAO");
        this.name = name;
        cg_bad = new();
    endfunction

    function void sample();
        cg_bad.sample();
    endfunction

    function void print();
        $display("%s, length %0d, address %0d", name, length, address);
    endfunction

endclass
```

MikePopoloski, 2023-03-09:
I was able to look at this more and I'm still not sure what isn't working for you. I ran slang with the --ast-json option on your example and I see the covergroup in the class:
```
 "name": "",
              "kind": "CovergroupType",
              "addr": 1941056014744,
              "members": [
                {
                  "name": "",
                  "kind": "CovergroupBody",
                  "addr": 1941055993528,
                  "members": [
                    {
                      "name": "option",
                      "kind": "ClassProperty",
                      "addr": 1941056011184,
                      "type": "struct{string name;int weight;int goal;string comment;int at_least;int auto_bin_max;int cross_num_print_missing;bit detect_overlap;bit per_instance;bit get_inst_coverage;}packet.s$1",
                      "lifetime": "Automatic",
                      "visibility": "Public"
                    },
                    {
                      "name": "type_option",
                      "kind": "ClassProperty",
                      "addr": 1941055979432,
                      "type": "struct{int weight;int goal;string comment;bit strobe;bit merge_instances;bit distribute_first;}packet.s$2",
                      "lifetime": "Static",
                      "visibility": "Public"
                    },
                    {
                      "name": "pippo",
                      "kind": "Coverpoint",
                      "addr": 1941056015248,
                      "members": [
                        {
                          "name": "option",
                          "kind": "ClassProperty",
                          "addr": 1941055983408,
                          "type": "struct{int weight;int goal;string comment;int at_least;int auto_bin_max;bit detect_overlap;}packet.pippo.s$3",
                          "lifetime": "Automatic",
                          "visibility": "Public"
                        },
                        {
                          "name": "type_option",
                          "kind": "ClassProperty",
                          "addr": 1941055985128,
                          "type": "struct{int weight;int goal;string comment;}packet.pippo.s$4",
                          "lifetime": "Static",
                          "visibility": "Public"
                        },
                        {
                          "name": "a",
                          "kind": "CoverageBin",
                          "addr": 1941056020000,
                          "binsKind": "Bins",
                          "isArray": false,
                          "isWildcard": false,
                          "isDefault": false,
                          "isDefaultSequence": false,
                          "values": [
                            {
                              "kind": "Conversion",
                              "type": "logic[5:0]",
                              "operand": {
                                "kind": "IntegerLiteral",
                                "type": "bit[5:0]",
                                "value": "6'b0",
                                "constant": "6'b0"
                              },
                              "constant": "6'b0"
                            }
                          ]
[... 102 lines cut ...]
                              "kind": "Conversion",
                              "type": "logic[1:0]",
                              "operand": {
                                "kind": "Conversion",
                                "type": "logic signed[31:0]",
                                "operand": {
                                  "kind": "IntegerLiteral",
                                  "type": "int",
                                  "value": "2",
                                  "constant": "2"
                                },
                                "constant": "2"
                              },
                              "constant": "2'b10"
                            }
                          ]
                        }
                      ]
                    }
                  ]
                }
              ]
            },
```
Can you explain more what's missing?

MikePopoloski, 2023-03-25:
Closing due to lack of response. Feel free to reopen if you can provide more info.

closed 2023-03-25 by MikePopoloski