MikePopoloski/slang #725: Cannot find virtual Interface's InstanceSymbol

HungMingWu, 2023-02-16:
**Describe the bug**
I am not sure whether it is a bug
Here is my test verilog code
``` verilog
interface synbus(input logic clk);
endinterface

typedef virtual synbus vi;
```

It create two InstanceSymbols, but only one in the elaboration tree..
But this one in elaboration tree is invalid, is it ok?

MikePopoloski, 2023-03-09:
There is no real instance in your example. There is a fake instance for the virtual interface type to reference, and there is the "invalid" instance of the interface which is created as a default case because nothing ever used that interface, so this lets slang examine the body of such an instance without it actually existing in the design.

MikePopoloski, 2023-03-25:
Closing this due to lack of response. Feel free to reopen if you have more questions.

closed 2023-03-25 by MikePopoloski