MikePopoloski/slang #727: internal compiler error

KingkingLiu, 2023-03-09:
internal compiler error: Assertion 'conns.size() == names.size()' failed
  in file /global/tools/freeware/slang/slang/source/ast/symbols/InstanceSymbols.cpp, line 890
  function: void slang::ast::UninstantiatedDefSymbol::serializeTo(slang::ast::ASTSerializer&) const

Build is succeeded but it's wrong when I add the option "--ast-json" to get the ast.

KingkingLiu, 2023-03-09:
Can I choose to remove this assetion ? or it will cause the error of compiler

MikePopoloski, 2023-03-09:
No, you can't remove the assert, it's telling you that there is a bug in the compiler. There's no way to know what that bug is though because you didn't provide an example snippet of code that reproduces the issue. Please provide steps to reproduce (the default issue template asks you to do this)

MikePopoloski, 2023-03-12:
Going to close this for now; feel free to reopen if you can provide more info.

closed 2023-03-12 by MikePopoloski