MikePopoloski/slang #728: crashes calling SourceManager::assignText

ichigensanokotowari, 2023-03-12:
I am 100% sure this is my fault, but I get a crash calling any overload of assignText. I suspect I am screwing something up when building the static lib. Essentially, I am 

0) using cmake gui to configure as static lib build
1) generating the SLN
2) rebuilding the SLN and building the INSTALL project to install the headers/libs

Then from my test project:

0) set include directory to C:\slang_install\include
1) set lib dirtectory to C:\slang_install\lib
2) add svlang.lib and fmtd.lib to the linker input settings

When I run, any attempt to call assignText dies horribly because
SourceLocation includedFrom
is invalid. So at this point, I am suspecting the library isn't built properly, there is a header/lib mismatch, I'm somehow not using it properly, or there is something not getting set up for the function call. Digging in a bit deeper, here is my test

`
	slang::SourceLocation loc;
	slang::string_view path = "";
	slang::string_view code =
		R"(
			module M;
			    typedef enum int { FOO = 1, BAR = 2, BAZ = 3 } test_t;

			    function void foo(int i, output r);
				endfunction
			endmodule
		)";

	slang::SourceBuffer buffer = sm.assignText(path, code, loc);
`
When stepping assignText. the location goes NULL. I'm not an x86 expert, but here is my sad attempt to work out what is going on

`
00007FF73D5D3ADE  lea         rax,[rbp+420h]  
00007FF73D5D3AE5  lea         rcx,[code]  
00007FF73D5D3AEC  mov         rdi,rax  
00007FF73D5D3AEF  mov         rsi,rcx  
00007FF73D5D3AF2  mov         ecx,10h  
00007FF73D5D3AF7  rep movs    byte ptr [rdi],byte ptr [rsi]  
00007FF73D5D3AF9  lea         rax,[rbp+3F0h]  
00007FF73D5D3B00  lea         rcx,[path]  
00007FF73D5D3B07  mov         rdi,rax  
00007FF73D5D3B0A  mov         rsi,rcx  
00007FF73D5D3B0D  mov         ecx,10h  
00007FF73D5D3B12  rep movs    byte ptr [rdi],byte ptr [rsi]  
00007FF73D5D3B14  mov         rax,qword ptr [loc]  
00007FF73D5D3B1B  mov         qword ptr [rsp+20h],rax  
00007FF73D5D3B20  lea         r9,[rbp+420h]  
00007FF73D5D3B27  lea         r8,[rbp+3F0h]  
00007FF73D5D3B2E  lea         rdx,[buffer]  
00007FF73D5D3B35  mov         rcx,qword ptr [sm]  
00007FF73D5D3B3C  call        slang::SourceManager::assignText (07FF73D5414BBh) 
`

RCX is the this pointer, which I think makes sense. Return type has a user defined constructor, and so it's pointer goes in rdx. Cool so far. Next I see both path and code copied to the stack, and their new stack addresses passed in r8 and r9 respectively. That looks correct. But now is where things go off the rails

`
00007FF73D5D3B14  mov         rax,qword ptr [loc]  
00007FF73D5D3B1B  mov         qword ptr [rsp+20h],rax  
`

rax is 0, which is what then gets written to the stack pointer. So it looks like we're getting the value of, not address of SourceLocation, which is then treated like a pointer inside the function.  Which is why I was initially suspecting the lib wasn't built correctly, or there was some header/lib mismatch.

So now I am wondering is there some unlisted build/setup step required to make it work? Or what else in the setup I might have missed, that everyone else using the lib seems to have gotten right

MikePopoloski, 2023-03-12:
Yes, I would guess that you have an ABI mismatch between the slang library you built and your own application. In general, includes+lib is not sufficient to link to a library -- you also need compile flags and link flags to be compatible. Probably you built slang in debug mode but have not carried across the `DEBUG` define. One of the things that define does is change the size of SourceLocation to contain extra debugging info, so your application's view of the structure and slang's view of the structure will differ. cmake handles automatically setting flags correctly when you link to targets but if you're doing it manually you have to know to seek them out and set them correctly.

ichigensanokotowari, 2023-03-12:
oh no! The second you said that, it was instantly understandable what the issue was. I didn't bother to check if the size of the struct was different with different defines. But of course the lib thinks it's > 64 bits and my test app thinks it's 64 bits, hence copying to the stack vs expecting an address on the stack. Thank you so much for your answer, and it all seems to be working now

closed 2023-03-13 by MikePopoloski