MikePopoloski/slang #736: Option to create executable that does not have dependency on shared objects of the OS

idjordje, 2023-03-30:
**Is your feature request related to a problem? Please describe.**
Current build option require fairly current OS distribution and tools versions. This prevents usage in environments that that are not easy to migrate to those revisions. Building a self sufficient (statically linked) executable) would allow building on an isolated system and distributing (plain copy) to older systems.

**Describe the solution you'd like**
Build a statically linked executable that does not depend on various dynamically linked .so libs

**Describe alternatives you've considered**
None

**Additional context**

MikePopoloski, 2023-03-30:
I'm open to this change but don't have the bandwidth to work on it myself. Feel free to open a PR with what you'd like to see.

sumn255, 2023-03-30:
I have the same requirement too.Now I modify CMakeLists.txt locally, don't know how to add a build option @idjordje 
in tools/CMakeLists.txt:line 8
-target_link_libraries(slang_driver PRIVATE slang::slang fmt::fmt)
+target_link_libraries(slang_driver PRIVATE slang::slang fmt::fmt -static)
line 13
-target_link_libraries(rewriter PRIVATE slang::slang)
+target_link_libraries(rewriter PRIVATE slang::slang -static)
in tools/tidy/CMakeLists.txt: line 6
-target_link_libraries(slang_tidy PRIVATE slang::slang fmt::fmt)
+target_link_libraries(slang_tidy PRIVATE slang::slang fmt::fmt -static)

furkanusta, 2023-03-30:
You can build statically, without any modifications
```cmake
cmake 
  -DBUILD_SHARED_LIBS=OFF 
  -DCMAKE_EXE_LINKER_FLAGS=-static 
  -DCMAKE_FIND_LIBRARY_SUFFIXES=.a
  ../
```

If you already have `fmt` in your system you might also need `-DCMAKE_DISABLE_FIND_PACKAGE_fmt=ON` so that it is pulled via FetchContent

MikePopoloski, 2023-04-04:
If this is as simple as setting some cmake command line flags then we can probably close this issue? @idjordje can you confirm?

idjordje, 2023-04-04:
I’ve tried suggested command line switches and they produce statically
linked executable that can be copied to other Unix systems.

@mikepopolski your call if you want to modify documentation saying “hey btw
use these additional switches” or modify the make files to have one clean
switch for all files. Seems executables in tools directory are built
separately from main slang executable

-Ivan

idjordje, 2023-04-04:
Btw if you decide on one clean switch for all executables I am fine with
providing you with PR

-Ivan

MikePopoloski, 2023-04-05:
Sorry, what exactly are you asking for with "one clean switch"? As far as I know, the CMAKE_ arguments you pass to cmake apply to all targets that get built, so I'm not sure why you're having trouble with the slang executable vs the tools.

In general there's an almost unlimited number of flags someone might want to build with to customize the build, so I'm hesitant to enshrine this particular one unless a large number of people need exactly this. If you want something more convenient for your local build you should look into using a CMakeUserPresets file, which can let you wrap up flags like this in a locally maintainable file.

idjordje, 2023-04-06:
Presets will work, you can close this issue.

 

-Ivan

closed 2023-04-08 by MikePopoloski