MikePopoloski/slang #744: Warnings for unused ports when type is inout

jrudess, 2023-04-17:
The following code-example mimics the 'scl' pin of an i2c interface.  The initial block in the top module is just example stimulus for two i2c master's driving scl to 0 at staggered times.

A few 'unused port' warnings trigger which I think should not have fired because of tranif1 statements which reference the port.   Ignore the unused-but-set-net warnings -- I didn't improve the example enough to add the missing sampling logic, and those warnings don't fire for the real-world code base.

```
slangtest133.sv:4:11: warning: unused port signal 'sig' [-Wunused-port]
    inout sig
          ^
slangtest133.sv:10:16: warning: unused port signal 'scl' [-Wunused-port]
    inout wire scl,
               ^
slangtest133.sv:14:10: warning: net 'scl_local' is driven but never referenced [-Wunused-but-set-net]
    wire scl_local;
         ^
slangtest133.sv:27:10: warning: net 'scl' is driven but never referenced [-Wunused-but-set-net]
    wire scl;
         ^
```
```Systemverilog
module pad(
    input en,
    input val,
    inout sig
);
    assign sig = en ? val : 1'bz;
endmodule

module i2c(
    inout wire scl,
    input scl_en,
    input scl_drive
);
    wire scl_local;

    pullup (scl_local);
    tranif1(scl, scl_local, 1'b1);

    pad pad_scl(
        .en (scl_en),
        .val (scl_drive),
        .sig (scl_local)
    );
endmodule

module top;
    wire scl;
    logic scl_en1, scl_en2;
    logic scl_drive1, scl_drive2;

    initial begin
        scl_en1 = 1'b0;
        scl_drive1 = '1;
        scl_en2 = 1'b0;
        scl_drive2 = '1;
        #100ns;
        scl_en1 = 1'b1;
        scl_drive1 = 1'b0;
        #50ns;
        scl_en2 = 1'b1;
        scl_drive2 = 1'b0;
    end

    i2c master1(
        .scl_en    (scl_en1),
        .scl_drive (scl_drive1),
        .*
    );
    i2c master2(
        .scl_en    (scl_en2),
        .scl_drive (scl_drive2),
        .*
    );
endmodule
```

jrudess, 2023-04-19:
Yeah, I was thinking I need inout to preserve some x-propagation, but not for that specific tiny module example.  Interestingly, if I do change it to output, then I get a warning that it's being coerced back to inout because of the tranif1 statement.

```
slangtest133.sv:22:15: warning: output net port 'scl_local' coerced to 'inout' [-Wport-coercion]
        .sig (scl_local)
              ^~~~~~~~~
slangtest133.sv:17:18: note: driven here
    tranif1(scl, scl_local, 1'b1);
                 ^~~~~~~~~
```

MikePopoloski, 2023-04-22:
Ah yeah, the rules about warning for port coercion don't care whether you actually read from the output port inside the module, only that you assign to it from the outside. Anyway, I believe I have fixed the warnings involved here.

closed 2023-04-22 by MikePopoloski