MikePopoloski/slang #745: silly segfault in pyslang

tdp2110, 2023-04-19:
**To Reproduce**

Consider

```
import pyslang

c = pyslang.Compilation()
c.addSyntaxTree(None)
```

This segfaults for me.

**Additional context**

The issue is in the pybinding [here](https://github.com/MikePopoloski/slang/blob/master/bindings/python/compilation.cpp#L100), which routes to [`Compilation::addCompilation`](https://github.com/MikePopoloski/slang/blob/master/source/ast/Compilation.cpp#L179), which does no null checking. 

As a fix, a null check could be added to the pybinding, which I'd be happy to do. You'd probably want to throw an exception, but I don't know which types you like.

Additional info: I was using this project for the first time, and happened to call `pyslang.SyntaxTree.fromFile("wrong-file-path.txt")`, which returned `None`. I then fed that to a compilation, segfaulted, and wondered if my `pip` install of `pyslang` was horked, which it was not. It also had no debug symbols, so the backtrace wasn't particularly helpful.

Thanks for the great work!

MikePopoloski, 2023-04-21:
Yeah, it's a simple fix but I'm happy to add it. Probably I'll add it to the addSyntaxTree method in the C++ -- I think that pybind11 was translate C++ exceptions into an appropriate Python exception.

MikePopoloski, 2023-04-22:
Fixed in 50aba881db9f397dc35a304b83d52aef8f5290ee

closed 2023-04-22 by MikePopoloski