MikePopoloski/slang #748: top-level module has unconnected interface port

MahmoudKMaarouf, 2023-04-28:
**Describe the bug**
Passing an interface of specific modport gives an error
But it compiles and synthesizes just fine 

**To Reproduce**

```
interface some_if;
logic a;
logic b;
modport master (input a, output b);
modport slave (output a, input b);
endinterface; 
```

```
module dut_dummy(
  some_if.master my_if,
  output c
);
assign my_if.b = 0;
assign c = my_if.a;
endmodule
```

```
module tb;
some_if my_if();
dut_dummy dut (.my_if(my_if.master), output c);
initial begin
  my_if.a = 1;
end
endmodule
```

**Additional context**
See picture attached
[image]

MikePopoloski, 2023-04-28:
I am confused by your example; the `dut_dummy` is not the top level module, the `tb` module is. Are you not actually including the tb in your compilation arguments, or are you specifically overriding it for some reason?

MahmoudKMaarouf, 2023-04-28:
For context, I am using the [VSCode SystemVerilog Extension](https://marketplace.visualstudio.com/items?itemName=mshr-h.VerilogHDL) with Slang.

When I switch between sv files, I would expect 0 errors on every file. 

I would have expected arguments -I or -y to work when specifying the tb path. But that did not work. 

The argument to fix the issue above would be to use -v {tb_file} . That would fix the issue with the design code. But then I would get the following warning:
warning: no top-level modules found in design [-Wmissing-top]

MikePopoloski, 2023-04-28:
Well I don't know exactly how the extension is invoking slang, but it sounds like it's treating each file like it's part of a normal elaboration flow, instead of all the files in your project together being part of the elaboration flow. Either it should include all the files from your project, or if it runs on each file separately it should probably run with `--lint-only` which turns off a bunch of checking that depends on a full project being elaborated. You shouldn't get false positives then, though you won't get 100% checking of all possible errors either.

MahmoudKMaarouf, 2023-04-28:
Please note that is there is another issue:

```
some_if another_if();
some_dut dut(.an_if(another_if.master));
```
Will NOT give an error

```
some_if another_if[3]();
some_dut dut(.an_if(another_if.master));
```
WILL give an error despite some_dut accepting an interface some_if of size [3] 

`Error: could not resolve hierarchical path name 'master' 
`

With 
`--lint-only` and 
`--ignore-unknown-modules`
options.

MikePopoloski, 2023-08-13:
Both issues mentioned in this thread are now resolved.

closed 2023-08-13 by MikePopoloski