MikePopoloski/slang #760: Compile error without BOOST

renau, 2023-05-21:
There is a compile error on top of the tree slang when BOOST is disabled.

This works file:
```
cmake ../ 
make
```

This fails:
```
cmake  -DSLANG_USE_BOOST=OFF ../
make
```

The error shown when doing `make VERBOSE=1`
```
cd /home/renau/projs/slang/build2/source && /usr/bin/c++ -DSLANG_STATIC_DEFINE -DSLANG_USE_MIMALLOC -I/home/renau/projs/slang/source/../include -I/home/renau/projs/slang/build2/source -I/home/renau/projs/slang/build2/_deps/fmt-src/include -isystem /home/renau/projs/slang/source/../external -isystem /home/renau/projs/slang/build2/_deps/mimalloc-src/include -O3 -DNDEBUG -std=c++20 -fvisibility=hidden -fvisibility-inlines-hidden -MD -MT source/CMakeFiles/slang_slang.dir/driver/Driver.cpp.o -MF CMakeFiles/slang_slang.dir/driver/Driver.cpp.o.d -o CMakeFiles/slang_slang.dir/driver/Driver.cpp.o -c /home/renau/projs/slang/source/driver/Driver.cpp
In file included from /usr/include/c++/11/bits/stl_algobase.h:64,
                 from /usr/include/c++/11/memory:63,
                 from /home/renau/projs/slang/source/../include/slang/ast/Compilation.h:10,
                 from /home/renau/projs/slang/source/../include/slang/driver/Driver.h:11,
                 from /home/renau/projs/slang/source/driver/Driver.cpp:9:
/usr/include/c++/11/bits/stl_pair.h: In instantiation of ‘struct std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode>’:
/usr/include/c++/11/ext/aligned_buffer.h:91:28:   required from ‘struct __gnu_cxx::__aligned_buffer<std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode> >’
/usr/include/c++/11/bits/hashtable_policy.h:234:43:   required from ‘struct std::__detail::_Hash_node_value_base<std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode> >’
/usr/include/c++/11/bits/hashtable_policy.h:268:12:   required from ‘struct std::__detail::_Hash_node_value<std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode>, true>’
/usr/include/c++/11/bits/hashtable_policy.h:277:12:   required from ‘struct std::__detail::_Hash_node<std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode>, true>’
/usr/include/c++/11/bits/hashtable_policy.h:1812:13:   required from ‘struct std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode>, true> > >’
/usr/include/c++/11/bits/hashtable_policy.h:811:13:   required from ‘struct std::__detail::_Insert_base<slang::ast::InstancePath::Entry, std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode>, std::allocator<std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode> >, std::__detail::_Select1st, std::equal_to<slang::ast::InstancePath::Entry>, slang::hash<slang::ast::InstancePath::Entry, void>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >’
/usr/include/c++/11/bits/hashtable_policy.h:1004:12:   required from ‘struct std::__detail::_Insert<slang::ast::InstancePath::Entry, std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode>, std::allocator<std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode> >, std::__detail::_Select1st, std::equal_to<slang::ast::InstancePath::Entry>, slang::hash<slang::ast::InstancePath::Entry, void>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true>, false>’
/usr/include/c++/11/bits/hashtable.h:180:11:   required from ‘class std::_Hashtable<slang::ast::InstancePath::Entry, std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode>, std::allocator<std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode> >, std::__detail::_Select1st, std::equal_to<slang::ast::InstancePath::Entry>, slang::hash<slang::ast::InstancePath::Entry, void>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true> >’
/usr/include/c++/11/bits/unordered_map.h:105:18:   required from ‘class std::unordered_map<slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode, slang::hash<slang::ast::InstancePath::Entry, void>, std::equal_to<slang::ast::InstancePath::Entry>, std::allocator<std::pair<const slang::ast::InstancePath::Entry, slang::ast::HierarchyOverrideNode> > >’
/home/renau/projs/slang/source/../include/slang/ast/Compilation.h:176:63:   required from here
/usr/include/c++/11/bits/stl_pair.h:218:11: error: ‘std::pair<_T1, _T2>::second’ has incomplete type
  218 |       _T2 second;                ///< The second member
      |           ^~~~~~
```

Using gcc 11.3.0

renau, 2023-05-21:
Trying different gcc, It compiles fails with gcc  11.3.0 but it compiles fine with  12.2.0

MikePopoloski, 2023-05-23:
Fixed by caa4dccc679abb315eeecfd380839526b2833c97

closed 2023-05-23 by MikePopoloski