MikePopoloski/slang #764: input cmake --build build in Ubuntu occurred error

jrainzzz, 2023-05-25:
**Describe the bug**
/home/zhang/slang/slang/source/ast/Statements.cpp:1597:46:   required from here
/usr/include/c++/10/bits/stl_uninitialized.h:137:72: error: static assertion failed: result type must be constructible from value type of input range
  137 | static_assert(is_constructible<_ValueType2, decltype(*__first)>::value,

make[2]: *** [source/CMakeFiles/slang_slang.dir/build.make:1396：source/CMakeFiles/slang_slang.dir/ast/Statements.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:1078：source/CMakeFiles/slang_slang.dir/all] 错误 2
make: *** [Makefile:163：all] 错误 2

How should I solve this problem？
thanks！

[image]

MikePopoloski, 2023-05-31:
Closing this for now. Feel free to reopen if you want to add more information.

jrainzzz, 2023-05-31:
ok,i will add more information tomorrow,thanks

 

Mr。Zhang
***@***.***

jrainzzz, 2023-06-01:
[image]
[image]
[image]
[image]
### then i continued to input commands.(cmake --build build)
[image]
### Then, when running to 79%, there was an error message when building Statements. cpp. o
[image]
[image]
[error_message.txt](https://github.com/MikePopoloski/slang/files/11621743/error_message.txt)

MikePopoloski, 2023-06-01:
Hmm, weird. I can build without issue with an almost identical setup. The only difference I can see is that my system has GCC 10.4.0 whereas you have 10.3.0. There's no straightforward way for me to downgrade, can you upgrade your GCC install and see if that fixes the issue?

jrainzzz, 2023-06-02:
I upgraded the gcc version to 10.4 but still encountered the same error

MikePopoloski, 2023-06-09:
Since there's been no updates I'm going to close this. Feel free to reopen if you have additional information we can use to debug.

closed 2023-06-09 by MikePopoloski