MikePopoloski/slang #768: Q - slang vs. pyslang code compatibility - feature (or a bug?)

svenka3, 2023-05-28:
Below code fails to compile on slang (code is faulty as per LRM, but a simulator allows). 

```verilog
class child_c;
  parameter NUM_BITS = 8;
endclass : child_c

class ex_c;
  child_c child_c_0;
  bit [child_c_0.NUM_BITS - 1 : 0] bit_vec;
  bit [8 - 1 : 0] bit_vec_1;
endclass : ex_c
```

I am able to parse it via pyslang though and flag it as:

```log
PySlint: Violation: [COMPAT_XMR_IN_WIDTH]:  A class property/variable is declared with XMR ' . ' in its type/width This is beyond LRM and not compatible across simulators:  Use scope resolution than hierarchical reference
  bit [child_c_0.NUM_BITS - 1 : 0]
```

Now, my question is - is this guaranteed/designed to be like this? Meaning, slang may fail, but pyslang will allow API access (for limited cases I suppose, not for a total random junk text). 

Thanks

MikePopoloski, 2023-05-30:
It depends on how you are using pyslang. Pyslang exposes the slang library API -- you are free to use the parts of it that you want and ignore the parts that you don't. The slang tool reports all diagnostics that it encounters, but if you don't use the relevant methods when using the API then not all diagnostics will be issued. For example, if you are only using the SyntaxTree API to parse files but aren't using the AST API (via the Compilation class) then you will only see syntax related diagnostics, not semantic / type checking diagnostics.

svenka3, 2023-05-30:
Thanks for the explanation, makes lot of sense. Am using SyntaxTree API in this case.

closed 2023-05-30 by svenka3