MikePopoloski/slang #770: Q - Python API to check for file wide items

svenka3, 2023-05-28:
Refer to:

https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#no-tabs

Maybe a dumb Q - but, what's the best approach to add such rules like:

1. No tabs
2. Max char per line
3. Max lines per file
etc.

Should I loop through root.members and rawText or something? 

Thanks in advance for any guidance.

MikePopoloski, 2023-05-30:
You can do it that way, yes. Visiting all of the tokens and trivia in depth first search order of a syntax tree is guaranteed to produce the original file (you can see the code in SyntaxPrinter.cpp for example) and so you should be able to enforce rules there. It might be even more straightforward though to just work at the Lexer level -- create a Lexer object with a source file and then call lex() on it until all tokens have been consumed, and check each token as it comes out to see things like whitespace and whatever else.

svenka3, 2023-05-30:
OK thanks, will try sometime soon.

closed 2023-05-30 by svenka3