MikePopoloski/slang #772: Statically Override Parameters as preprocessing

Pflyg, 2023-06-05:
Hello, I wanted to use slang to statically override parameters in some top-level modules, to create several versions of the original SV file, just with different parameters.

I have the following verilog file:
```
module mux #(parameter n = 4) (
  input [n-1:0] in,
  input [$clog2(n)-1:0] select,
  input clk,
  output reg out
);
  wire [0:$clog2(n)-1] reversed = select;
  assign out = in[reversed];
endmodule
```

And I tried using `slang -G n=2 -E mux.vl`, expecting the following output:
```
module mux (
  input [3:0] in,
  input [1:0] select,
  input clk,
  output reg out
);
  wire [0:1] reversed = select;
  assign out = in[reversed];
endmodule
```
But I just got the original file.
Do you have any guidance if and how this could be achieved using either the command line, python, or cpp libraries?

MikePopoloski, 2023-06-05:
It is possible to build such a tool using the library (python or cpp) but such functionality doesn't currently exist. The thing you're missing is the ability to go from the fully elaborated AST back to source code. Such a tool might be a decent amount of work to build depending on how pretty you want the output / how close you want it to match the original source. For example, what happens if you have a module definition that is used more than once with different parameter values assigned? You'll need to duplicate the definition in the output, so it's not as simple as reprinting the input source with minor modifications.

Pflyg, 2023-06-05:
Thanks for the quick response! I was looking at the JSON AST output which transformed back to source (after using the -G parameter) would probably suffice for my usage case, so I was hoping I could go directly from that back to source using that or the internal representation. Original formatting would be nice, but not strictly required

MikePopoloski, 2023-07-10:
FWIW [this project](https://github.com/pulp-platform/svase) seems like it might be what you want (it's built on top of slang).

MikePopoloski, 2023-07-29:
Closing this out since I believe the question has been answered.

closed 2023-07-29 by MikePopoloski