MikePopoloski/slang #779: Mux in netlist path

renau, 2023-06-26:
**Describe the bug**

The variables in the if/else are not used in the  netlist.

**To Reproduce**

The following module generates a dot file (slang-netlist) where the `sel` is not used for the `mux.f` but clearly the mux selector is part of the sources.

```
module mux(input a, input b, input sel, output reg f);

always @(*) begin
  if (sel == 1'b0) begin
    f = a;
  end else begin
    f = b;
  end
end

endmodule
```

MikePopoloski, 2023-06-26:
@jameshanlon

jameshanlon, 2023-06-26:
I'll take a look at this. The problem is that dependencies are not propagated from the variables appearing in the condition to the ones they control (ie the targets of any assignments). I should also add a note to the `slang-netlist` readme that the tool is currently a work in progress!

PR #781 (2023-07-01): [slang-netlist] Fix connectivity of procedural conditional statements 
This fixes #779 by propagating variable references appearning in conditional statement expressions to be dependencies of output variables of the procedural block. This same approach also needs to be applied to other statements such as case.
commit 6f80c8dc (2023-07-01): [slang-netlist] Fix connectivity of procedural conditional statements  (#781)

Fixes #779.
closed 2023-07-01 by MikePopoloski