MikePopoloski/slang #782: Question: how Driver.diagEngine initialized with sourceManager?

albertethon, 2023-07-03:
Is the `sourceManager` properly initialized in the member initial list while initializing the `diagEngine` in `Driver.cpp:39`?

```cpp
Driver::Driver() : diagEngine(sourceManager) {
    diagClient = std::make_shared<TextDiagnosticClient>();
    diagEngine.addClient(diagClient);
}
```

What factors influence the initialization of `diagEngine.clients`?
I have observed that, on certain occasions, the driver initializes clients at an incorrect memory address. 
I successfully initialized a driver in a simple project and obtained the correct clients. 
However, when dealing with circt/slang-fronted, something went a wrong way.
For detailed information regarding `diagEngine.clients`, please refer to the following links:
https://discourse.llvm.org/t/is-circt-able-to-analyze-verilog/66281/23?u=albertethon

MikePopoloski, 2023-07-03:
I responded to that thread last week. I'll copy my response here as well:

> I have to assume you’re using different build configurations for slang vs llvm. The DEBUG flag is just one example that can affect ABI, though it’s the one people usually run into first. Generally that’s why you want to either build them in the same cmake invocation (by pulling in the sources via add_subdirectory) or by using the cmake package via find_package.

If you think you're building everything correctly you'll need to provide much more detailed steps for reproducing the problem.

albertethon, 2023-07-07:
Thanks, I import slang in my new project and it was running correctly. After checking the `--verbose` log I found that it was some flags that `slang-fronted` doesn't declare. It run correctly after these macro flags declared in cmake. Your assumption was right.

closed 2023-07-07 by albertethon