MikePopoloski/slang #786: reference to ‘text_style’ is ambiguous

albertethon, 2023-07-10:
**Describe the bug**
I build the slang v3.0 and it complains reference to ‘text_style’ is ambiguous, slang uses `fmt::v9::text_style` and `fmt::v10::text_style` existed in `/usr/local/include/fmt/color.h`
However these bug disappeared in master branch, which slang version is 3.0.207

**To Reproduce**
```bash
git checkout v3.0
cmake -B build -DCMAKE_BUILD_TYPE=Debug
cmake --build build -j
```
**Detail**
```
/home/wissy/szy/slang/source/util/OS.cpp:119:22: error: reference to ‘text_style’ is ambiguous
  119 | void OS::print(const fmt::text_style& style, string_view text) {
      |                      ^~~
In file included from /home/wissy/szy/slang/source/util/OS.cpp:8:
/home/wissy/szy/slang/source/../include/slang/util/OS.h:18:7: note: candidates are: ‘class fmt::v9::text_style’
   18 | class text_style;
      |       ^~~~~~~~~~
In file included from /home/wissy/szy/slang/source/util/OS.cpp:19:
/usr/local/include/fmt/color.h:232:7: note:                 ‘class fmt::v10::text_style’
  232 | class text_style {
      |       ^~~~~~~~~~
```

MikePopoloski, 2023-07-10:
slang is always built against a specific version of the fmt library. If you're relying on the system installed version then that version needs to match. I'm a little surprised cmake didn't complain about finding the wrong version though when you configured the project.

albertethon, 2023-07-31:
May I ask how to solve this question? cmake didn't complain when configured the project. What should I do if I want to use v3.0 slang?

albertethon, 2023-08-01:
Thanks, I tried the first one and it works well.

closed 2023-08-02 by MikePopoloski