MikePopoloski/slang #792: slang-netlist assertion on bus expression in ports

heiner-bauer, 2023-07-24:
**Describe the bug**
Using a (functionally redundant) bus expression in ports triggers the assertion in netlist::SplitVariables::isIntersectingSelect https://github.com/MikePopoloski/slang/blob/4d10a392ada5500ee551e3fccb8e5b0118a440f0/tools/netlist/include/SplitVariables.h#L50

**To Reproduce**
The issue occurs due to the expression inside of i_nop.in_i:
```verilog
module test (
             input [1:0] in_i,
             output [1:0] out_o
             );

   wire [1:0]             in_s;

   assign in_s = in_i;

   nop i_nop(
               .in_i(in_s[1:0]), // ok: in_s, in_i, {in_i[1], in_i[0]}
               .out_o(out_o)
               );
endmodule

module nop (
             input [1:0]  in_i,
             output [1:0] out_o
             );

   // individual bits access; ok: out_o = in_i;
   assign out_o[0] = in_i[0];
   assign out_o[1] = in_i[1];
endmodule
```

**Additional context**
I was trying to analyze paths in a large, generated Verilog netlist to see if slang-netlist would be more efficient than the Verilator based netlist-paths.

Btw., would it be possible to use slang as a rewriter to remove such "redundant" expressions (i.e. in_i[LEN-1:0] with in_i) from a Verilog netlist ?

MikePopoloski, 2023-07-25:
@jameshanlon FYI

To answer your last question, yes, slang can be used as a rewriter. There isn't a declarative way of doing it though; you have to link slang as a library and write the rewriting logic based on how you want it to work, though there is a SyntaxRewriter class to help you out a bit.

jameshanlon, 2023-07-25:
@heiner-bauer Thanks for the bug report. The short answer is that currently `SplitVariables` doesn't handle ranges or unions. I'm working on refactoring it to do so by operating on bit positions rather than symbols, indices and ranges. I'll reply on this issue as soon as I have something.

PR #826 (2023-09-24): [slang-netlist] Refactor handling of variable references with selectors
This patch refactors slang-netlist's handling of variable references with selectors. This handling is used to determine if two distinct variable references have data dependencies. Previously, a simplistic scheme was used that only handled array element selects and struct member selects. The new scheme calculates the bit range for any combination of selectors by recursively inspecting the data types. One omission is that it does not (yet) handle descending ranges, eg `[0:3]`.

Fixes #792.
closed 2023-09-24 by MikePopoloski