MikePopoloski/slang #793: slang-netlist assertion on port name collision with unused modules

heiner-bauer, 2023-07-24:
**Describe the bug**
Having unused module definitions with identical port names (e.g. large cell libraries) triggers the assertion in netlist::Netlist::addVariableDeclaration https://github.com/MikePopoloski/slang/blob/92261b9dbd4d8c6e555031db41ea6dc0d97c4cfd/tools/netlist/include/Netlist.h#L287

 The unused modules appear to end up in a global namespace which leads to collisions when calling `lookupVariable`.
I guess there should be a preprocessing to trim unused module definitions or ignore them while walking the netlist AST ?

**To Reproduce**
The modules cell_b and cell_c are unused and have identical port names
```verilog
module test (
             input i1,
             input i2,

             output o1
             );

   cell_a i_cell_a(
                   .d1(i1),
                   .d2(i2),
                   .c(o1)
                   );
endmodule

module cell_a(
              input  d1,
              input  d2,

              output c);

   assign c = d1 + d2;
endmodule

// unused
module cell_b(
              input  a,
              input  b,

              output z);

   assign z = a || b;
endmodule

// unused
module cell_c(
              input  a,
              input  b,

              output z);

   assign z = (!a) && b;
endmodule
```

heiner-bauer, 2023-07-24:
The assertion only triggers when manually specifying the toplevel module with `slang-netlist --top test test.v`

MikePopoloski, 2023-07-25:
@jameshanlon FYI

PR #807 (2023-08-08): [slang-netlist] Exclude unused modules from netlist
Fix #793

Using `--top` causes unused (ie uninstanced) modules to lose their symbol names.

Since it is not necessary to include unused modules in a netlist, this patch skips them during the AST traversal.
closed 2023-08-08 by MikePopoloski