MikePopoloski/slang #794: configure fails to find the pre-installed unordered_dense package

yurivict, 2023-07-26:
**Describe the bug**
```
CMake Error at external/CMakeLists.txt:85 (install):
  install TARGETS given target "unordered_dense" which does not exist.
Call Stack (most recent call first):
  CMakeLists.txt:214 (include)
```

**To Reproduce**
Regular configure with ```-DFETCHCONTENT_FULLY_DISCONNECTED:BOOL=ON```.

Version: 3.0
clang-15
FreeBSD 13.2

MikePopoloski, 2023-07-26:
Thanks for the report. The logic has changed quite a bit in master compared to the last release (for example, unordered_dense isn't even used anymore); can you retry with the current version and see how that goes? I assume you'll run into similar problems with other packages but at least it will make sense with the current code.

Also please include your cmake version. The find_package fallback logic only works with cmake >=3.24, so if you don't have that you can also start by trying that.

yurivict, 2023-07-26:
cmake-core-3.26.1

yurivict, 2023-07-26:
```
===>  Configuring for slangverilog-3.0.286
===>  Performing out-of-source build
/bin/mkdir -p /usr/ports/cad/slang/work/.build
-- CMake version: 3.26.1
-- slang version: 3.0.0+0
-- The CXX compiler identification is Clang 15.0.7
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/libexec/ccache/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Python: /usr/local/bin/python3.9 (found version "3.9.17") found components: Interpreter 
-- Build STATIC library as: svlang
-- Using remote fmt library
-- Found system boost 1.82.0 at /usr/local/include
-- Found system mimalloc version: 
-- Using remote Catch2 library
CMake Error at external/CMakeLists.txt:145 (install):
  install TARGETS given target "fmt" which does not exist.
Call Stack (most recent call first):
  CMakeLists.txt:169 (include)

-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Configuring incomplete, errors occurred!
*** Error code 1

```

MikePopoloski, 2023-07-28:
Ok, it's unfortunate that cmake doesn't give a better error here but at least you should know how to fix the problem. I can see about adding some messaging in the cmake file if the find_package fails and also fetchcontent downloading is turned off.

MikePopoloski, 2023-07-29:
6c40f7f6d5f7e9a53b13393977496b9763f05198 adds some checking to the cmake build for this scenario.

closed 2023-07-29 by MikePopoloski

yurivict, 2023-07-29:
This is orthogonal to this issue, but libfmt-10 causes a lot of build failures in various other packages, and it isn't easy to upgrade libfmt to 10 for this reason. Since slang in libfmt-10 only, it can't be ported at the moment.

MikePopoloski, 2023-07-30:
libfmt is very easy to statically link to; it's almost a header-only library. You could statically link to it while keeping the other dependencies shared to avoid interfering with the older system-installed libfmt.