MikePopoloski/slang #795: Modport on arrays of SVIs in port assignments are not supported

Cyberwizzard, 2023-07-26:
**Describe the bug**
Modport on arrays of SVIs in port assignments are not supported. A modport on a single SVI in port assignments seems to work (as I do not get a syntax error on those).

This syntax is used in commercial RTL where afaik all our tools for both ASIC and FPGA flows have parsed it correctly.
I searched the LRM for proof whether it is or is not supposed to be supported but I can not find definitive proof either way so I am opening this ticket.

**To Reproduce**
```
interface MyInt( ... );
   ...
   modport mo_some( .... );
endinterface

module B (
   MyInt.mo_some interfaces [1:0]  // Array of modport typed interfaces
);
  ...
endmodule

module A ();
   MyInt interfaces [1:0] ();

   B (.interfaces( interfaces.mo_some );  // Throws syntax error about the .mo_some
endmodule
```

**Additional context**
Dropping the `.mo_some` fixes the syntax error but causes warning in other programs about directionality being unknown at the time of compilation (which is why we want to provide the modport on the port assignment in the first place).

MikePopoloski, 2023-07-26:
Yeah, this was recently reported in #748 as well. The LRM does not say anything about this being supported but all the major tools do allow it so I'll plan on adding it to slang as well.

MikePopoloski, 2023-08-13:
Support for this has been added.

closed 2023-08-13 by MikePopoloski