MikePopoloski/slang #796: Getting comments from syntax tree nodes

Cyberwizzard, 2023-07-26:
**Is your feature request related to a problem? Please describe.**
I'm looking into slang to parse, compile and elaborate RTL for documentation purposes (using Python).

To this end I need to obtain a list of parameters and ports, their types and names and in case of parameters; their values.

Each parameter and port will have a description in the RTL which I need to grab as well to combine with the syntax tree of the design. As far as I can establish, all comments will have been discarded from the syntax tree as they are not part of the design.

**Describe the solution you'd like**
Is there a way to get comment nodes in the SyntaxTree? Or get comments some other way once I know the names of all parameters and ports?

**Describe alternatives you've considered**
Using regex on the RTL source to obtain the info if I can get to the line numbers, but there has to be a more elegant way to get to this data.

MikePopoloski, 2023-07-26:
The comments (and other non-parseable constructs like whitespace and preprocessor directives) are stored as "trivia" alongside the next real token that is encountered. Tokens are the leaves of the SyntaxNode tree, so you can find the nearest token you care about for a particular syntax node and then look at its (leading) trivia to find comments.

Some explanatory documentation here: https://sv-lang.com/parsing.html#syntax-nodes
An example using the Python bindings (C++ will be essentially the same idea): https://github.com/MikePopoloski/pyslang/blob/master/tests/test_extractcomments.py

Cyberwizzard, 2023-07-27:
Perfect! The test_extractcomments.py does exactly what I wanted, thanks!

closed 2023-07-27 by Cyberwizzard