MikePopoloski/slang #800: --suppress-warnings no longer supports paths

jrudess, 2023-07-27:
In earlier versions, I passed paths to both --suppress-warnings and --suppress-macro-warnings.  This allowed easy suppression for all code contained in 3rd party libraries that exist outside of the repo being tested.

With HEAD, I see the following warnings now, and the warnings contained within the libraries are no longer being suppressed:
```
warning: --suppress-warnings path '/my/path/to/uvm': Is a directory
warning: --suppress-macro-warnings path "/my/path/to/uvm': Is a directory
```
Is the intention that only individual files are supported?

MikePopoloski, 2023-07-27:
All file related inputs to slang were changed to take a pattern (or list of patterns) in the format used by the LRM for declaring library maps. To achieve the same effect as before you want something like `/path/to/uvm/...` or `/path/to/uvm/.../*.sv` . I haven't had a chance to update the docs yet. The suppress-warnings feature is one of the few places where this is not backwards compatible, but I felt that long term the consistency was worth the break.

jrudess, 2023-07-28:
Running into some weird issues where after changing the UVM suppress-macro pattern from /path/to/uvm to /path/to/uvm/*/src/*/*.svh that I get compile errors about undefined macros, like all those svh files have been completely removed from the compile instead of just had their warnings suppressed.  Will see if I can reproduce in a standalone UVM environment.

jrudess, 2023-07-28:
While trying to reproduce, I think there is a threading or file-access race conditions.  Sometimes I'm getting the same compile errors that I see in my full-environment, but other times I get the following exception:

```
terminate called after throwing an instance of 'slang::assert::AssertionException'
  what():  Assertion 'inserted' failed
  in file /home/jason/repos/slang/source/text/SourceManager.cpp, line 488
  function: slang::SourceBuffer slang::SourceManager::cacheBuffer(std::filesystem::__cxx11::path&&, std::string&&, slang::SourceLocation, const slang::SourceLibrary*, slang::SmallVector<char>&&)

Abort (core dumped)
```

Command line (run from UVM tar-ball root dir):
```
slang src/uvm_pkg.sv test.sv -Isrc -Wextra --suppress-macro-warnings src/*/*/*.svh
```

Test case:
```Systemverilog
`include "uvm_macros.svh"

package my_package;
    import uvm_pkg::*;

    class my_agent extends uvm_agent;

        string s;

        `uvm_component_utils_begin(my_agent)
            `uvm_field_string(s, UVM_ALL_ON)
        `uvm_component_utils_end

        function new(string name = "my_agent", uvm_component parent = null);
            super.new(name, parent);
        endfunction

    endclass

endpackage
```

MikePopoloski, 2023-07-28:
Uh oh, that's not good. I'll try to reproduce; in the mean time, does using `--threads=1` make it go away?

jrudess, 2023-07-28:
I'm not able to reproduce the exception with ```--threads=1```.

And then if I remove --suppress-macro-warnings from the command line the errors go away as well.

MikePopoloski, 2023-07-28:
Ok, I see what's happening here. It's not actually a threading bug, it's just that that argument slightly changes the order that arguments are processed. The `src/*/*/*.svh` string is being interpreted by your shell and expanded before slang even sees it, which means the first result of that expansion is used as the --suppress-macro-warnings argument and the rest are passed as source files to slang. Putting single quotes around that argument to suppress the shell behavior, or using the form `--suppress-macro-warnings=src/*/*/*.svh` would also work.

There is an actual bug here, which is that if you pass a header file to slang as a source file (via a positional argument) and then also include that header file from some other source file you get that assertion in debug mode. I'll fix that but it's pretty degenerate behavior anyway since probably nobody ever actually wants to do that.

jrudess, 2023-07-28:
Ah, yeah.  Shell expansion completely skipped my mind.   Adding just an equal-sign was not sufficient (tcsh).  I needed to add single quotes around all cases with wildcards.

jrudess, 2023-07-28:
I take back my comment about the equal-sign not being sufficient.  I'm getting some inconsistent errors on compile.  Most of the time it passes with only warnings, but sometimes there are errors.  The errors are no longer reproducible when I use --threads=1

MikePopoloski, 2023-07-28:
Can you give more detail about what exactly is happening? I'm not able to reproduce any problems with the example from earlier.

jrudess, 2023-07-28:
The RTL base I'm working in has real horrible `include practices and relies heavily on the filelist ordering for everything to work correctly.   About 1 in 3 Slang runs has failures like this:

```systemverilog
src/rtl/common/dut_define.vm:311:10 error: redefinition of 'dut_p_cover'
property dut_p_cover(seq, clk, rst);
         ^
```

But with  --threads=1, this doesn't happen.  I have a few ideas on how to create a stand-alone example that can reproduce, and will try it out this next week and submit a new issue if I can create a good case to use.

MikePopoloski, 2023-07-28:
Ok yeah, sounds like a different problem. Certainly that's caused by the parallel parsing, includes will be reached in a non-deterministic order, but I'm having a hard time thinking of ways that could cause differing behavior.

MikePopoloski, 2023-07-28:
Also do you not use `--single-unit`? That basically turns off multithreaded parsing anyway...

jrudess, 2023-07-28:
Yeah, I have ```--single-unit``` enabled for all my environments.  I also noticed that I did not get the non-deterministic behavior of errors until after I added wildcards into the suppress options.  I'm not sure how those would be related though.

MikePopoloski, 2023-07-28:
I don't see how it could be related actually. If you remove that option you don't get the problem anymore even over multiple runs?

jrudess, 2023-07-28:
My mistake.  I just removed all suppress options and see the same non-deterministic failures at a rate of about 1 in 3 runs.    So high confidence this a new threading issue to track down and nothing to do with suppress options.

MikePopoloski, 2023-07-28:
Ok, great. I'll close this then and let you post a new issue (and I'll think a bit more how the threading could be causing this in the meantime).

closed 2023-07-28 by MikePopoloski

MikePopoloski, 2023-08-01:
I figured out what I was doing wrong with the load order. Should be fixed in 4d84bc9932a26710d2563fdda59fdc690967a072