MikePopoloski/slang #802: Reference to undefined param in another param default is not flagged if that param is overridden at instantiation

jrudess, 2023-07-29:
The below example has a compile error behavior difference between slang and VCS.  I don't think that slang is wrong, per se, but thought it was an interesting case to see what you think.

VCS always gives a compile error that the p1 argument passed to $clog2 is not defined.   Slang gives that same error, but only when p2 is not overridden at module instantiation as it is in the below example.

```Systemverilog
module dut #(
    // p1 = 0,
    parameter p2 = $clog2(p1)
);
endmodule

module top;
   dut #(.p2(3)) dut_i();
endmodule
```

jrudess, 2023-07-29:
Should also mention, that the reason I commented p1 out was because it was being flagged as an unused parameter even though it it is referenced in the p2 default.

MikePopoloski, 2023-08-02:
I remember when thinking about this originally that I was worried someone was setting their default param values to something invalid as a way to specifically get compile errors when forgetting to override them (as a poor man's version of non-defaulted parameters). My preference would be to validate the defaults though, so I'm leaning towards fixing this, although I did test with VCS and it did not provide an error -- maybe my version is too old?

jrudess, 2023-08-02:
Was running with VCS 2022.06-SP2.  It definitely could be a newer change or related to compile options (e.g. when using partitioned compile I've run into very large differences in reported errors in the past vs regular compile modes).

closed 2024-04-27 by MikePopoloski