MikePopoloski/slang #809: Request: New -Wunassigned-but-read-variable

jrudess, 2023-08-17:
-Wunassigned-variable helps catch variables that might not be used, but this is very often a huge set of warnings in large projects with lots of legacy code that was never properly cleaned up.  So I generally need to disable this specific warning.

But, what I'd really like to know is whether any of those unassigned-variables are being read, e.g. -Wunassigned-but-read-variable, since that would mean there is likely a real bug to go investigate.

Would a warning like this be feasible?

MikePopoloski, 2023-08-17:
In theory that's what Wunassigned-variable already does. If the variable is completely unused (neither read nor written) then you get Wunused-variable. If it's read but never set you get Wunassigned-variable, and if it's set but never read you get Wunused-but-set-variable. If you find places where that logic isn't holding let me know.

jrudess, 2023-08-17:
Thanks, I missed the distinction -- will review my results and reopen if I find anything.

closed 2023-08-17 by jrudess