MikePopoloski/slang #810: -Werror=empty-stmt flags errors but doesn't display them

jrudess, 2023-08-17:
I have an environment where there are no empty-stmt warnings being flagged and the run passes with 152 warnings.  But when I rerun with ```-Werror=empty-stmt``` then the run fails but doesn't print any errors:

```
Build failed: 324 errors, 152 warnings
```

Do you have any recommendations on how to track this down?  The compile does have many unused modules, but a simple test-case with unused modules does print the errors, and I haven't found a reproducer yet.

jrudess, 2023-08-17:
Ah, of course.  This is happening because of --suppress-warnings.  The warnings exist in files that are in the suppression list.   When the warning is converted to an error, it looks like the warning still gets suppressed.

MikePopoloski, 2023-08-17:
That's not the intended behavior. I'll fix it.

MikePopoloski, 2023-08-17:
Fixed in 4b9e3f1845dfdb50255aefcd731897d7ff1fc4c4

closed 2023-08-17 by MikePopoloski