MikePopoloski/slang #812: "mimalloc-static" fails CMake configuration when slang v4.0 is used as a library

suzizecat, 2023-08-20:
**Describe the bug**
When using slang as dependency (using CMake's "FetchContent" method), I get the following errors right after cmake configuration:

```
...
[cmake] -- Using remote mimalloc library
...
[cmake] -- Configuring done
[cmake] CMake Error in build/_deps/mimalloc-src/CMakeLists.txt:
[cmake]   Target "mimalloc-static" INTERFACE_INCLUDE_DIRECTORIES property contains
[cmake]   path:
[cmake] 
[cmake]     "/home/julien/Projets/C++/slang-lsp-tools/build/_deps/slang-src/"
[cmake] 
[cmake]   which is prefixed in the build directory.
[cmake] 
[cmake] 
[cmake] CMake Error in build/_deps/mimalloc-src/CMakeLists.txt:
[cmake]   Target "mimalloc-static" INTERFACE_INCLUDE_DIRECTORIES property contains
[cmake]   path:
[cmake] 
[cmake]     "/home/julien/Projets/C++/slang-lsp-tools/build/_deps/slang-src/"
[cmake] 
[cmake]   which is prefixed in the build directory.Target "mimalloc-static"
[cmake]   INTERFACE_INCLUDE_DIRECTORIES property contains path:
[cmake] 
[cmake]     "/home/julien/Projets/C++/slang-lsp-tools/build/_deps/slang-src/"
[cmake] 
[cmake]   which is prefixed in the source directory.
[cmake] 
[cmake] 
[cmake] -- Generating done
[cmake] CMake Generate step failed.  Build files cannot be regenerated correctly.
```

**To Reproduce**
Have a sample project using slang as a library, with `v4.0` . This is *not* reproducible with tag `v3.0`. 
Using 
```console
$ cmake --version
cmake version 3.22.5

CMake suite maintained and supported by Kitware (kitware.com/cmake).
``` 
```cmake
FetchContent_Declare( slang
  GIT_REPOSITORY https://github.com/MikePopoloski/slang.git
  GIT_TAG v4.0
  GIT_SHALLOW ON)

FetchContent_MakeAvailable(slang)
```

**Additional context**
For reference, the CMake output that seems related to slang:
```
[cmake] -- Build type: Debug
[cmake] -- slang version: 4.0.0+007e5ab
[cmake] -- Found Python: /usr/bin/python3.10 (found version "3.10.12") found components: Interpreter 
[cmake] -- Build STATIC library as: svlang
[cmake] -- Using remote fmt library
[cmake] -- Using vendored boost_unordered header
[cmake] -- 
[cmake] -- Skip collecting memory on program exit (MI_SKIP_COLLECT_ON_EXIT=ON)
[cmake] -- 
[cmake] -- Library base name: mimalloc-debug
[cmake] -- Version          : 2.1
[cmake] -- Build type       : debug
[cmake] -- C Compiler       : /usr/bin/gcc-12
[cmake] -- Compiler flags   : -Wall;-Wextra;-Wno-unknown-pragmas;-fvisibility=hidden;-Wstrict-prototypes;-ftls-model=initial-exec
[cmake] -- Compiler defines : MI_SKIP_COLLECT_ON_EXIT=1
[cmake] -- Link libraries   : /usr/lib/x86_64-linux-gnu/libpthread.a;/usr/lib/x86_64-linux-gnu/librt.a
[cmake] -- Build targets    : static
[cmake] -- 
[cmake] -- Using remote mimalloc library
[cmake] -- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
[cmake] -- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
[cmake] -- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
[cmake] -- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
[cmake] -- Performing Test COMPILER_HAS_DEPRECATED_ATTR
[cmake] -- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
[cmake] -- Configuring done
```

suzizecat, 2023-08-20:
As a workaround, setting 
```cmake
option(SLANG_USE_MIMALLOC ""  OFF)
```
before making slang available solves the issue but probably with some drawbacks that I am not aware of.

MikePopoloski, 2023-08-20:
There is an example project using slang as a library via FetchContent here: https://github.com/MikePopoloski/slang_subproject_example

The CI build has been passing. Can you figure out how your case differs?

suzizecat, 2023-08-21:
I am based upon this very example project. The only difference I can see is that I explicitly request the v4.0 version.

I see that the CI of the example project has been ran 4 months ago. And I think that the breaking change (for me) is more recent than that. Could you try to force a CI run (of the example project) to check if it is still OK ?

MikePopoloski, 2023-08-21:
The build runs daily; see for example last night's: https://github.com/MikePopoloski/slang_subproject_example/actions/runs/5918759152/job/16048368150

I can try changing it to request the tag specifically and see if that does anything.

suzizecat, 2023-08-22:
Indeed I hadn't looked in the right place... 
Would you happen to know which version of CMake is used in this CI process ?

suzizecat, 2023-08-22:
I found where the issue lied, and indeed it is not due to slang but my own issue.
I was using 
```cmake
include_directories(${uri_SOURCE_DIR})
```

Prior to make available Slang, where `${uri_SOURCE_DIR}` is the source dir of a header file retrieved by `FetchContent`. 
Thus adding a prefixed path to the include directories which was checked afterward for... whatever cmake reason, I guess... ? 

Anyway, I moved this `include_directories` directive later in the process and used `$<BUILD_INTERFACE:${uri_SOURCE_DIR}>` and the problem seems to be solved.

closed 2023-08-22 by suzizecat