MikePopoloski/slang #813: [feature] Allow union to struct assignments with matching formats for -Wimplicit-conv

jrudess, 2023-08-21:
00ae5276a77f834e96866075a54e5c08bbbcd620 really helped remove a lot of 'bogus' struct assignment warnings in RTL (particularly when signals are being handed off from one IP to another iP that uses different definitions for the same struct.

One case that remains is when the RHS of the assignment is a union instead of a struct.   Would it be possible to extend this change to support the union case as well?

```Systemverilog
slangtest143.sv:31:20: warning: implicit conversion from 's4_t' to 's1_t' [-Wimplicit-conv]
        assign s1b = s4;
                   ^ ~~
```
```Systemverilog
typedef struct packed {
    logic a;
    logic [1:0] b;
    logic c;
} s1_t;

typedef struct packed {
    logic a;
    logic [1:0] b;
    logic c;
} s2_t;

typedef struct packed {
    logic a;
    logic [1:0] b;
    logic c;
} s3_t;

typedef union packed {
    s2_t s2;
    s3_t s3;
} s4_t;

module top;
    s1_t s1a, s1b;
    s2_t s2;
    s4_t s4;

    always_comb begin
        assign s1a = s2;
        assign s1b = s4;
    end
endmodule
```

jrudess, 2023-08-21:
Decided that this is really just poor code.  If a union is being used, the sub-structs should all have different formats.  If they have different formats then a mismatch should fire.   If all the sub-structs are identical, then the logic is just weird.

Requiring an explicit typecast is probably appropriate in the code here.

MikePopoloski, 2023-09-02:
I think there's some value in loosening the warnings here; assigning one of the constituents *to* a packed union should basically always be safe, for example.

MikePopoloski, 2023-12-18:
Done in cbe87d2a912379ae1492bc5f1aefc5983721e076

closed 2023-12-18 by MikePopoloski