MikePopoloski/slang #815: Get Buffer(ID) by file name

suzizecat, 2023-08-24:
**Is your feature request related to a problem? Please describe.**
As of today (except if I missed an API function), there is no way to easily get the BufferID of an already read file.

An application example is in the case of a LSP server. If I want to lookup a symbol by position, I will the LSP will provide a file path and a position in this file. In order to fetch information in the AST, I will need to look for the SyntaxNode related to the target position.

In order to do so, It will be relevant to use a `slang::SourceLocation` to hold the target location, but I then need the BufferID linked to my file path.

**Describe the solution you'd like**
I would like a function `BufferID getBufferByPath(std::filesystem::path fp)` (or equivalent) in the source manager class.

**Describe alternatives you've considered**
I came up with a solution that iterates through all buffers of the source manager and search for the right filename.
```cpp
const slang::SourceManager* sm = compilation->getSourceManager();
BufferID target_id;
for (BufferID id : sm->getAllBuffers())
{
    if (sm->getFullPath(id).filename() == "my_file.sv")
    {
        target_id = id;
        break;
    }
}

int target_position = 1234;
slang::SourceLocation sl(target_id, target_position);
```

MikePopoloski, 2023-08-26:
This seems reasonable. I can add this.

suzizecat, 2026-03-11:
I have also updated my method since then. 
I store the CST root of the file in a dedicated object and use a visitor to lookup references of previously detected symbols. 

Thanks for your time considering this !