MikePopoloski/slang #822: Is there a convenient function to get the token?

xunfeilong, 2023-09-18:
Hello, Mr.Popoloski. Sorry to bother you. I am Hailong Sun, working on Moore's translation, which uses slang to convert a Verilog/SV file into an AST, and then convert the AST into IR. This can be viewed at https://github.com/fabianschuiki/circt/tree/slang-frontend. When I handle the ternary operator, I find AST's right node is different from others, like an assignment in the Initial block. Is there a convenient function to get the specified kinds, rather than using the recursive method to know what kind is?

MikePopoloski, 2023-09-18:
Do you have an example of what you're looking for? In what way is the AST's right node different from others? I also don't know what you mean by the "specified kinds". Kinds of AST nodes? Tokens? SyntaxNodes?

xunfeilong, 2023-09-19:
[https://github.com/xunfeilong/cmakeTutorial/blob/main/assign.png](url)
[https://github.com/xunfeilong/cmakeTutorial/blob/main/ConditionalOp.png](url)
        The image links above are two examples I wrote. The help I am looking for is when I deal with the right node (or token?). I must call .as<slang::ast::ConversionExpression> to get the inner data layer by layer, and I can not skip any one layer. For example, in the assign.png, when I want to get the literal 1: 
Firstly, x = something.right.as<slang::ast::ConversionExpression>; 
Secondly, y = x.operand.as<slang::ast::ConversionExpression>; 
Lastly, z = y.operand.as<slang::ast::IntegerLiteral>, only in this way can I get the literal 1, and can not ignore any step, otherwise, the error message will be thrown. Is there another method that can avoid processing layer by layer and get a kind, like IntegerLiteral, directly? And I do not know what is the difference between tokens and nodes. I looked at AllSyntax.cpp. I found that the assignment's child-count is 4, like assign a = 1; why is the child-count 4? Another question is that I can not seem to get the literal of a variable by the address.

MikePopoloski, 2023-09-24:
There is some documentation here https://sv-lang.com/parsing.html which covers tokens and nodes (these are not terms made up for slang but are something that is common to basically all parsers). As for your question, you generally need to be aware of conversions that happen so there isn't some built in way to ignore them. You can visit them generically by using the ASTVisitor helper which may help make your code simpler.

If you don't care about type information you can operate on the concrete syntax tree (i.e. the SyntaxTree class and nodes) instead of using the abstract syntax tree which is what you're currently looking at. The concrete tree will match what you see typed in the source code much more closely.

xunfeilong, 2023-09-25:
I get it. Thank you very much for your advice. I read the documentation you mentioned. I think I know what the problem is.

closed 2023-09-26 by MikePopoloski