MikePopoloski/slang #823: Incorrect resolution of array position with dynamic range selects

jameshanlon, 2023-09-18:
**Describe the bug**
When selecting a range from a packed array using an increasing or decreasing incremental range select where the base index is non constant, the resolved range is incorrect.

**To Reproduce**
```
$ cat bug.sv
module m (input int a);
  logic [3:0] [1:0] foo;
  always_comb begin
    foo[a+:1][1] = 0;
    foo[a-:1][1] = 0;
  end
endmodule

$ ./bin/slang bug.sv
Top level design units:
    m

bug.sv:4:15: error: cannot refer to element 1 of 'logic[0:0][1:0]' [-Windex-oob]
    foo[a+:1][1] = 0;
              ^
bug.sv:5:15: error: cannot refer to element 1 of 'logic[3:3][1:0]' [-Windex-oob]
    foo[a-:1][1] = 0;
              ^

Build failed: 2 errors, 0 warnings
```
I would expect the outer array dimension to be `[3:0]` as defined.

MikePopoloski, 2023-09-18:
Hmm, you're taking an array slice of size 1 and I guess expecting it to decompose to a non-array since it's exactly of size 1? I can look more at what the LRM says here but your expectation that the outer dimension be [3:0] doesn't make sense to me since you explicitly took a slice of size 1, not 4. Or are you thinking the slice applies to the right-hand bounds [1:0]? If so I believe that's wrong, indexing is left-to-right.

It's also worth noting that I think taking an index after a slice is technically not allowed by the LRM to begin with. They restrict the slice to be the final select in a chain of selects. I should probably update slang to enforce that restriction but it seems pretty artificial.

jameshanlon, 2023-09-18:
Perhaps this isue is a moot point if the LRM does not permit indexing after a slice.

Notwithstanding, I think the diagnostic is still confusing. Taking another example:
```
$ ./bin/slang bug-823.sv
Top level design units:
    m

bug-823.sv:4:15: error: cannot refer to element 15 of 'logic[2:0]' [-Windex-oob]
    foo[a+:3][15] = 0;
              ^~
bug-823.sv:5:15: error: cannot refer to element 1 of 'logic[31:29]' [-Windex-oob]
    foo[a-:3][1] = 0;
              ^

Build failed: 2 errors, 0 warnings
```
In this, it seems that both data types produced by the slices should be `logic[2:0]`. It looks like the data type ranges are based on `a` = 0.

I agree my previous comment is not quite right, where the data type after the slice should be 1 bit wide.

MikePopoloski, 2023-09-18:
Yeah, I suspect the question of what the returned bounds should be is partly why the construct is disallowed in the first place. For example, if someone has a convention that they declare all their vectors as 1-based, in ascending order: `logic [1:8]` they might be surprised to find that the result of the slice is suddenly zero based. slang's choice here is also confusing though, I'll give you that.

MikePopoloski, 2023-11-25:
Fixed in a31e602d5232d3699ce3d359fd6cc12b325aca8a

closed 2023-11-25 by MikePopoloski