MikePopoloski/slang #824: Multi-dimensional, typeless localparam assignment results in "assignment pattern target type cannot be deduced in this context"

alwilson, 2023-09-18:
**Describe the bug**
Assigning values to multi-dimensional localparam results in an error. Offending line appears to pass linting and compilation on vendor tools. Adding a type to the localparam declaration appears to pass. I'm not sure what LRM section this falls under or what is more valid here.

**To Reproduce**
```systemverilog
localparam     [1:0][7:0] VALUES_0 = '{1, 2};
localparam bit [1:0][7:0] VALUES_1 = '{1, 2};
```

```systemverilog
$ slang localparam_dimensions.sv 
localparam_dimensions.sv:1:38: error: assignment pattern target type cannot be deduced in this context
localparam     [1:0][7:0] VALUES_0 = '{1, 2};
                                     ^~~~~~~
warning: no top-level modules found in design [-Wmissing-top]

Build failed: 1 error, 1 warning

$ slang --version
slang version 4.0.40+cf39a993
```

**Additional context**
I don't believe I have anything non-standard here.

MikePopoloski, 2023-09-18:
This is running into the definition of "assignment-like contexts". The LRM says that a parameter assignment is only an assignment-like context if it has an explicit type declaration. It's a little ambiguous what that means, but the grammar says that an implicit data type (which is only allowed in specific places in the grammar) is:
```
implicit_data_type ::= [ signing ] { packed_dimension }
```
or in other words, having just a range or signing but no explicit data type is still considered an "implicit data type" and hence my reading is that it's not an assignment-like context.

There's no particularly good reason to prevent this from working though when you have the range declared, since the target type is in fact known, and indeed all the tools I tried it with allow it so I'll change slang to allow it as well.

MikePopoloski, 2023-09-24:
Fixed in e0d4122daeae7b69def13a26013ebe092d74b154

closed 2023-09-24 by MikePopoloski