MikePopoloski/slang #825: Wrong index variant on slang-reflect

Sustrak, 2023-09-19:
The following construct will cause a:
```
terminate called after throwing an instance of 'std::bad_variant_access'
  what():  std::get: wrong index for variant
Aborted (core dumped)
```

the construct:

```systemverilog
localparam [CORE_ID_BITS - 1:0] CORE_ID_0 [pkg::NR_CORES] = '{2, 3, 0, 0, 0}; 
localparam [CORE_ID_BITS - 1:0] CORE_ID_1 [pkg::NR_CORES] = '{4, 5, 0, 0, 0}; 
localparam [CORE_ID_BITS - 1:0] CORE_ID [2][pkg::NR_CORES] /* verilator public */ = '{CORE_ID_0, CORE_ID_1};  
```

MikePopoloski, 2024-01-20:
@Sustrak can you provide a PR to fix this?

Sustrak, 2024-02-12:
> @Sustrak can you provide a PR to fix this?

Sorry for taking that long, didn't had much time to work on it. Expect a PR during this week :smile:

PR #894 (2024-02-16): Reflect more complex local parameters
With this commit slang-reflect is able to reflect more complex local parameters. Also, it will throw an error if it doesn't know how to reflect the parameter instead of crashing.

Fixes #825
commit f904c154 (2024-02-16): Reflect more complex local parameters (#894)

* Fixes #825

With this commit slang-reflect is able to reflect more complex
local parameters. Also, it will throw an error if it doesn't know
how to reflect the parameter instead of crashing.
closed 2024-02-16 by MikePopoloski