MikePopoloski/slang #829: Undriven struct members do not trigger -Wunassigned-variable

jrudess, 2023-09-26:
Unsure if this is a bug report or a feature request.   The warnings that check for reading unassigned variables has a small gap if that variable is a member of a struct, but this might be intentional.

In the following example, only member 's.a' is driven and 's.b' is undriven.

```systemverilog
Top level design units:
    top

Build succeeded: 0 errors, 0 warnings
```
```systemverilog
typedef struct {
    logic a;
    logic b;
} s_t;

module top;
    s_t s;
    logic x, y, z;
    always_comb s.a = 1'b1;

    always_comb begin
        if (s.b) begin
            x = 1'b1;
        end else begin
            x = 1'b0;
        end
    end

    always_comb y = s.b;

    always_comb begin
        z = s.b ? 1'b1 : 1'b0;
    end

endmodule
```

MikePopoloski, 2023-09-26:
Yes, it's intentional. I can add a new warning set for struct members though.

jrudess, 2023-09-27:
I'm assuming that a full struct initialization would count as all members having a driver (s = '0).   Most code is written this way, it's just the occasional one-off module where the designer uses individual assign statements for each struct member that I've hit a few cases of missed drivers.   A new warning here would be great to help push finding the issue to earlier in the process.