MikePopoloski/slang #833: Make slang available on ConanCenter

learningstud, 2023-10-18:
**Is your feature request related to a problem? Please describe.**
For mere convenience.

**Describe the solution you'd like**
Make slang available on ConanCenter.

**Describe alternatives you've considered**
As suggested by the manual: fetch the source, build the source, and install system wide.

**Additional context**
No.

MikePopoloski, 2023-10-18:
I think this is probably an issue for https://github.com/conan-io/conan-center-index -- if you need fixes from me to make it smoother / easier I'm happy to help but I expect it should be straightforward already.

learningstud, 2023-10-20:
Yes, thanks for pointing out my erroneous attribution of liability. I have opened an issue at conan-center-index.

closed 2023-10-20 by learningstud