MikePopoloski/slang #838: Allow reading arguments from file

suzizecat, 2023-10-26:
**Is your feature request related to a problem? Please describe.**
Most tools allows reading arguments from a file (in order to save a consistent settings, generate the list of all files of a design...) which doesn't seems to be possible with slang.

**Describe the solution you'd like**
An argument (e.g. `-f`) allowing to read a file and use all lines that are not commented (usually using `#` or `//`) as an argument, treating newlines as a white space.

**Describe alternatives you've considered**
Given the requirement of comments, I cannot see something beside an external file parser calling back slang 

**Additional context**
For reference, `-f` argument from verilator : https://veripool.org/guide/latest/exe_verilator.html#cmdoption-0

MikePopoloski, 2023-10-26:
slang supports `-f` -- you can see a reference for all command line options here: https://sv-lang.com/command-line-ref.html

suzizecat, 2023-10-27:
Damn, and here I searched in the `--help` output for the `-f` and the `--exclude-ext` arguments. 
Is there a particular reason for those to not be shown in the help output or would you consider a pull request to add them ? 
(It won't be immediate as I do not have much time right now but this is definitely something I could do)

MikePopoloski, 2023-10-27:
Hmm, I see them when I run with --help?

```
  --exclude-ext <ext>                                 Exclude provided source files with these extensions
  -f <file-pattern>[,...]                             One or more command files containing additional program options. Paths in the file are considered relative to the current directory.
  -F <file-pattern>[,...]                             One or more command files containing additional program options. Paths in the file are considered relative to the file itself.
```

MikePopoloski, 2023-11-04:
Going to close this -- feel free to reopen if you can add additional context.

closed 2023-11-04 by MikePopoloski

suzizecat, 2023-11-09:
On the machine I've made the test with, I somehow had the version 0.8 as executable...
Sorry for the noise.