MikePopoloski/slang #842: Traverse the hierarchy and find all instance names

riggy2013, 2023-10-31:
Hi, Mike,

I tried to traverse the whole verilog/sv file to find all instance names for further processing.

What's the best way to walk through the whole hierarchy and find all HierarchyInstantiation or PrimitiveInstantiation? I didn't find a good way and write below code instead:

```
def check_members(sm, member):
  if not member:
    return
  if member.kind == pyslang.SyntaxKind.HierarchyInstantiation or \
    member.kind == pyslang.SyntaxKind.PrimitiveInstantiation:
    inst = member.instances[0]
    tok = inst.getFirstToken()
    ln = sm.getLineNumber(tok.location)
    if inst.decl != None:
      inst_name = inst.decl.name.value
    if not re.match(r'^i_|^i\d+_', inst_name):
      print('%d s/%s/%s/' % (ln, inst_name, inst_rep(inst_name.lower())))
  else:
    if hasattr(member, 'members'):
      for sub_memb in member.members:
        check_members(sm, sub_memb)
    if hasattr(member, 'items'):
      for sub_item in member.items:
        check_members(sm, sub_item)
    if hasattr(member, 'block'):
      check_members(sm, member.block)
    if hasattr(member, 'clause'):
      check_members(sm, member.clause)
    if hasattr(member, 'elseClause'):
      check_members(sm, member.elseClause)
```

BTW, any good reference document to describe the class structures? The current sv-lang document seems not updated. I have to use python debugger to check all members of class object.

thanks,

riggy2013

MikePopoloski, 2023-11-01:
You can use a visitor object -- the .visit() method on AST objects will traverse all child objects. You can look at the test_visitors.py test in the pyslang repo for an example.

The sv-lang.com docs should be up to date, but they only exist for C++. I'd like to have Python-specific docs but it's not something I have time to create.

MikePopoloski, 2023-11-04:
Closing -- feel free to reopen if you need additional clarification.

closed 2023-11-04 by MikePopoloski