MikePopoloski/slang #850: slang-netlist: Segmentation fault on adder example

alwilson, 2023-11-29:
**Describe the bug**
Running `slang-netlist` with the given adder example results in a segmentation fault.

**To Reproduce**
I cloned and ran the Quick Start build instructions without issue.

Version: `c29f3e9a` (HEAD -> master, origin/master, origin/HEAD) Fix netlist unittests broken by recent changes

adder.sv:
```systemverilog
module adder
  #(parameter p_width = 32)(
    input  logic [p_width-1:0] i_a,
    input  logic [p_width-1:0] i_b,
    output logic [p_width-1:0] o_sum,
    output logic               o_co
  );
  logic [p_width-1:0] sum;
  logic co;
  assign {co, sum} = i_a + i_b;
  assign o_sum = sum;
  assign o_co = co;
endmodule
```

failing command:
```bash
$ ./build/bin/slang-netlist adder.sv --from adder.i_a --to adder.o_sum
Top level design units:
    adder

Build succeeded: 0 errors, 0 warnings
Segmentation fault (core dumped)
```

GDB backtrace using cmake debug build (`-D CMAKE_BUILD_TYPE=Debug`):
```cpp
Program received signal SIGSEGV, Segmentation fault.
0x0000555555729df2 in slang::ast::Symbol::getParentScope (this=0x0) at /home/alex/git/slang/source/../include/slang/ast/Symbol.h:152
152         const Scope* getParentScope() const { return parentScope; }
(ins)(gdb) bt
#0  0x0000555555729df2 in slang::ast::Symbol::getParentScope (this=0x0) at /home/alex/git/slang/source/../include/slang/ast/Symbol.h:152
#1  0x0000555555a1cbac in slang::ast::getHierarchicalPathImpl (symbol=..., buffer=...) at /home/alex/git/slang/source/ast/Symbol.cpp:88
#2  0x0000555555a1d2ab in slang::ast::Symbol::getHierarchicalPath (this=0x0, result="") at /home/alex/git/slang/source/ast/Symbol.cpp:162
#3  0x00005555558873ca in operator() (__closure=0x200000330d8, sym=...) at /home/alex/git/slang/source/ast/Compilation.cpp:170
#4  0x00005555558bec47 in std::__invoke_impl<std::__cxx11::basic_string<char>, slang::ast::Compilation::Compilation(const slang::Bag&)::<lambda()>::<lambda(const slang::ast::Symbol&)>&, const slang::ast::Symbol&>(std::__invoke_other, struct {...} &) (__f=...) at /usr/include/c++/13.2.1/bits/invoke.h:61
#5  0x00005555558be67f in std::__invoke_r<std::__cxx11::basic_string<char>, slang::ast::Compilation::Compilation(const slang::Bag&)::<lambda()>::<lambda(const slang::ast::Symbol&)>&, const slang::ast::Symbol&>(struct {...} &) (__fn=...) at /usr/include/c++/13.2.1/bits/invoke.h:116
#6  0x00005555558be0ea in std::_Function_handler<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >(const slang::ast::Symbol&), slang::ast::Compilation::Compilation(const slang::Bag&)::<lambda()>::<lambda(const slang::ast::Symbol&)> >::_M_invoke(const std::_Any_data &, const slang::ast::Symbol &) (__functor=...,
    __args#0=...) at /usr/include/c++/13.2.1/bits/std_function.h:291
#7  0x000055555560ec5a in std::function<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > (slang::ast::Symbol const&)>::operator()(slang::ast::Symbol const&) const (this=0x200000330d8, __args#0=...) at /usr/include/c++/13.2.1/bits/std_function.h:591
#8  0x000055555560b7bd in slang::TextDiagnosticClient::report (this=0x200000330a0, diag=...) at /home/alex/git/slang/source/diagnostics/TextDiagnosticClient.cpp:76
#9  0x00005555555ccd59 in slang::DiagnosticEngine::issueImpl (this=0x7fffffffcb50, diagnostic=..., severity=slang::DiagnosticSeverity::Note)
    at /home/alex/git/slang/source/diagnostics/DiagnosticEngine.cpp:256
#10 0x00005555555cbcab in slang::DiagnosticEngine::issue (this=0x7fffffffcb50, diagnostic=...) at /home/alex/git/slang/source/diagnostics/DiagnosticEngine.cpp:190
#11 0x0000555555574d9e in reportPath (compilation=..., path=...) at /home/alex/git/slang/tools/netlist/netlist.cpp:132
#12 0x0000555555576166 in main (argc=6, argv=0x7fffffffd8f8) at /home/alex/git/slang/tools/netlist/netlist.cpp:260
```

**Additional context**
Pretty sure I tried this and it worked before, but it's been months since I tried it.

alwilson, 2023-11-29:
I checked out the `v4.0` tag, which worked fine, then I worked back a few commits from `master` and this seems to be the commit it breaks on https://github.com/MikePopoloski/slang/commit/e9a9243ec5e506795552abcb5c4b4e0b3e318d66

MikePopoloski, 2023-11-30:
Thanks for bisecting; I see the bug.

MikePopoloski, 2023-11-30:
Should be fixed in 079fffd0b18b7e7ab806d5674de0615731fe59d0

closed 2023-11-30 by MikePopoloski