MikePopoloski/slang #863: Nested Module Definition Not Reflected in slang --ast-json Output

yanggeorge, 2023-12-22:
**Is your feature request related to a problem? Please describe.**
I've encountered a problem when using the slang tool to compile and analyze SystemVerilog code with nested modules. Specifically, after compiling the following SystemVerilog code and using the slang --ast-json command to inspect the Abstract Syntax Tree (AST), I noticed that the AST representation does not seem to include the nested TestModule definition within the testModule's scope.
```systemverilog
module testModule(input logic CLK, output logic RST);
    logic clk;
    logic rst;

    module TestModule(input logic CLK, output logic RST);
        logic clk;
        logic rst;
    endmodule

endmodule
```

**Describe the solution you'd like**
I would like the slang --ast-json output to accurately reflect the nested module structure, showing the TestModule as part of the testModule's scope. Ideally, the AST should provide a clear and hierarchical representation of all module definitions, including nested modules, to facilitate better analysis and understanding of the code structure.

**Describe alternatives you've considered**
As an alternative, I've considered manually parsing the SystemVerilog code to track nested modules, but this approach is prone to errors and does not leverage the powerful analysis capabilities of slang. Another alternative might be using a different tool for AST generation, but I prefer to understand if slang can be enhanced or if I'm missing something in its current functionality.

**Additional context**
Below is the JSON output from the slang --ast-json command after compiling the provided SystemVerilog code. As you can see, the testModule's scope includes ports and internal variables but does not contain the nested TestModule definition. Understanding why the nested module's definition is not included in the AST representation is crucial for comprehensive analysis of complex SystemVerilog designs.
```json
{
  "name": "$root",
  "kind": "Root",
  "addr": 2199023598400,
  "members": [
    {
      "name": "",
      "kind": "CompilationUnit",
      "addr": 2199025901088
    },
    {
      "name": "testModule",
      "kind": "Instance",
      "addr": 2199025908848,
      "body": {
        "name": "testModule",
        "kind": "InstanceBody",
        "addr": 2199025907728,
        "members": [
          {
            "name": "CLK",
            "kind": "Port",
            "addr": 2199025908984,
            "type": "logic",
            "direction": "In",
            "internalSymbol": "2199025909160 CLK"
          },
          {
            "name": "CLK",
            "kind": "Variable",
            "addr": 2199025909160,
            "type": "logic",
            "lifetime": "Static"
          },
          {
            "name": "RST",
            "kind": "Port",
            "addr": 2199025909536,
            "type": "logic",
            "direction": "Out",
            "internalSymbol": "2199025909712 RST"
          },
          {
            "name": "RST",
            "kind": "Variable",
            "addr": 2199025909712,
            "type": "logic",
            "lifetime": "Static"
          },
          {
            "name": "clk",
            "kind": "Variable",
            "addr": 2199025908008,
            "type": "logic",
            "lifetime": "Static"
          },
          {
            "name": "rst",
            "kind": "Variable",
            "addr": 2199025908384,
            "type": "logic",
            "lifetime": "Static"
          }
        ],
        "definition": "testModule"
      },
      "connections": [
      ]
    }
  ]
}       
```
Thank you for your time and assistance.
Best regards

yanggeorge, 2023-12-23:
In slang, Definition class does not inherit from Symbol. And Scope.members() is only about symbols. So I cannot from members to find nested Definitions. 

But I find every definition has ref to a scope. It seems that the only way to find hierarchical relationship is to traverse the definitionMemory in compiltion instanse.

Why Scope class cannot contain a vector of definitions?

MikePopoloski, 2023-12-23:
The AST only includes *instances*, not definitions themselves. This is because the AST is supposed to represent the elaborated design tree, and a definition that is not instantiated is not part of the elaborated design tree. If you were to instantiate that nested module you should see it in the output.

yanggeorge, 2023-12-24:
@MikePopoloski  Thanks for your reply. I have got that AST don't include the unused definitions.

So It is hard to write a tidy check on all definitions, since Compilation class don't give such API  to access them.

MikePopoloski, 2023-12-24:
It wouldn't be that hard to add such an API to the Compilation, since it has that info internally. I can do that.

yanggeorge, 2023-12-24:
Thank you. I wish that it would not make conflict to your API design.

closed 2023-12-24 by yanggeorge

MikePopoloski, 2023-12-26:
Added in 05e147efa7c787d0e1c09049214b80076048c1b6