MikePopoloski/slang #866: Get a source range accessor in (Const)TokenOrSyntax

suzizecat, 2024-01-03:
**Is your feature request related to a problem? Please describe.**
I am working on an indexer that uses slang and allows me to lookup the references to a `Symbol` (given a `Symbol`, the list of `SourceRange` that actually refers to this symbol) and the definition of those references (Given a `SourceRange`, retrieve the `Symbol`).

Some references can actually be token (and not `SyntaxNode`), such as `.myPort` in a module instanciation which is the `NamedPortConnectionSyntax::name` token.
Therefore, I am working with `ConstTokenOrSyntax` as a stored type for the references within my data structures.

Each time I want to access to the source range of a reference, I must use something such as 

```cpp
// const ConstTokenOrSyntax& ref = ... 
SourceRange sr = ref.isNode() ? ref.node()->sourceRange() : ref.token().range();
```

**Describe the solution you'd like**
I'd prefer simply using
```cpp
SourceRange sr = ref.range(); 
```

**Describe alternatives you've considered**
I am using the first form everywhere.

**Additional context**
It may be interesting to check if there are other common attributes in syntax nodes and tokens that could be pushed to `(Const)TokenOrSyntax` but I haven't needed anything else yet.

MikePopoloski, 2024-01-05:
Sure, this seems straightforward.

MikePopoloski, 2024-01-20:
Added in 3eb51f0862c908368f467aec6fffc9f5ef794bed

closed 2024-01-20 by MikePopoloski