MikePopoloski/slang #876: Missing `endif not caught in included file

abradd, 2024-01-25:
**Describe the bug**

With the following code in file `slang_test.sv`:

``` verilog
`ifdef __slang__
`define SBY(x) 1
`else
`define SBY(x) x
`endif

module m;

  `ifdef FORMAL
    `include "slang_include.sv"
  `endif
    
endmodule
```

where `slang_include.sv` is given by:

``` verilog
`ifdef __slang__
`define TEST(x) 1
`else
`define TEST(x) x
`endif
```

Removing the `` `endif`` from `slang_include.sv` reports no errors. Removing either `` `endif`` in `slang_test.sv` will report an error for the missing `` `endif`` directive.

**To Reproduce**

Create the two files above and run:

``` sh
slang -Weverything slang_test.sv
```

jrudess, 2024-01-26:
Because +define+FORMAL is not on the slang command line, the pre-processor completely removes all code within the `ifdef FORMAL, so compile can't flag the error because the code doesn't exist now.

MikePopoloski, 2024-01-27:
Yes, that's right. slang won't even try to open the include file here so it doesn't matter what you put into it.

abradd, 2024-01-28:
> Because +define+FORMAL is not on the slang command line

Sorry about that, I was testing with and without `-DFORMAL`, but it seems I posted the one without.

If I run:

``` sh
slang -Weverything -DFORMAL slang_test.sv
```

after removing the `` `endif`` from `slang_include.sv` I still don't see an error. Is there something else I'm missing?

MikePopoloski, 2024-01-29:
Ok yeah, looks like there's still a bug here, specifically with include files.

MikePopoloski, 2024-02-08:
Fixed in 918f4f44a81b5e5aab56e09caf89a4c5a2ec7697

closed 2024-02-08 by MikePopoloski