MikePopoloski/slang #880: Fixed windows build with latest sources.

Romain-Piquois, 2024-01-29:
Ignore this message if there is no need for it. I managed to build the sources on Windows but I had to fix two issues.

1/ Inside SVIntHelpers.h, if the target is x64, #include <intrin.h> is not properly included.
I think it is due to #if defined(_MSC_VER) && defined(_M_IX86) at line 43

2/ Inside Debug.h , MSVC did not like the ( , ) for VA arguments to DebugMessage and InfoMessage call in the macro. 
I removed the bracket and simply used ',' (comma) character instead.

Everything builds successfully after that.

MikePopoloski, 2024-01-29:
I'm a little confused; I build on Windows locally and there's a CI job that builds for Windows as well. What are you doing differently?

Romain-Piquois, 2024-01-29:
Sorry. I hope I don't waste your time :-)

I followed the instruction given by your page : https://sv-lang.com/building.html

Those command were run using the windows shell (so not using WSL or the like ) .Python3 was already installed:
- added conan using : pip install conan.
- conan install . -b missing

Resulted in an error on my machine because I had no conan profile, so :
- conan profile detect
Then :
- conan install . -b missing (again but pass this time)
- cmake -B build (other cmake command did not ran properly and I am not very fluent in those tools)

After that, went to /slang/build/ and opened slang.sln (visual studio solution file).
And ran "Build Solution" in the IDE...

Then I had those two errors popping up. The first error resulted in not finding _addcarry_u64 and the like.
I noticed that #include <intrin.h> was indeed there, but disabled due to the false preprocessor condition in my IDE (Building as x64 by default)
`
#if defined(_MSC_VER) && defined(__x86_64__)
`

I just shamelessly fixed the issue by adding directly #include <intrin.h> without any consideration on my machine.
(Obviously it is a hack)

Then again fixed the other error. as described earlier.

My software : Microsoft Visual Studio Community 2022, V.17.4.4

Romain-Piquois, 2024-01-29:
Just adding here the result of conan profile detect, I have many different version of Visual Studio on my dev machine. (so it may have picked up another version ?)
`
detect_api: Found msvc 17

Detected profile:
[settings]
arch=x86_64
build_type=Release
compiler=msvc
compiler.cppstd=14
compiler.runtime=dynamic
compiler.version=193
os=Windows

WARN: This profile is a guess of your environment, please check it.
WARN: The output of this command is not guaranteed to be stable and can change in future Conan versions.
WARN: Use your own profile files for stability.
`

MikePopoloski, 2024-01-29:
If you don't use --preset with the cmake command then you aren't using the conan environment and it's doubtful things will work correctly, so I assume that's the problem. The conan install step tells you what prefix to use, or you can run `cmake --list-presets` to find it.

MikePopoloski, 2024-02-04:
Closing since there has been no followup.