MikePopoloski/slang #893: unsigned-conversion reports logic type but expected int unsigned

jrudess, 2024-02-14:
When compiling this example with -Wsign-conversion, a conversion to a 4-state logic is reported but all variables are 2-state.  I would expect this to be an 'int' to 'int unsigned' conversion.

```Systemverilog
class C;
      function f();
          int unsigned c, m;
          for (int b = 0; b < m; b++) begin
              int unsigned a = b / c;
          end
      endfunction
endclass
```
```
slangtest150.sv:5:32: warning: implicit conversion changes signedness from 'int' to 'logic[31:0]' [-Wsign-conversion]
              int unsigned a = b / c;
                               ^ ~
```

jrudess, 2024-02-14:
Here's a slightly different variant involving 2-state bit types.

```Systemverilog
class C;
      function f();
          int unsigned v[8];
          foreach (v[i]) begin
              bit [4:0] b = 5'd30 + 5'(i/4);
          end
      endfunction
endclass
```
```
slangtest151.sv:5:37: warning: implicit conversion changes signedness from 'logic signed[4:0]' to 'logic[4:0]' [-Wsign-conversion]
              bit [4:0] b = 5'd30 + 5'(i/4);
                                  ~ ^~~~~~~
```

MikePopoloski, 2024-02-14:
The result of the division operator is always a 4-state type, because purely 2-state inputs can still result in 'x (when dividing by zero). The relevant LRM section here is `11.3.4 Operations on logic (4-state) and bit (2-state) types`

You can observe the differences in behavior with a little test like this:
```
function automatic int foo;
    int n = 8, zero = 0;
    int div2 = n/zero + n;
    return div2;
endfunction

module m;
  int n = 8, zero = 0;
  initial $display($typename(n/zero));
  initial $display(foo());
endmodule
```
If int/int resulted in a 2-state result, foo() would return 8, but all sims I tried correctly return 0. Most sims report the correct typename as well, but VCS bizarrely reports `int` even though they return the correct result from foo.

jrudess, 2024-02-14:
Thanks, I had missed the exception for division operator.  I suspect VCS reports 'int' so that people stop reporting it as a bug to them like I just did!

closed 2024-02-14 by jrudess